procount/pinn #761: Home Assistant

unichargepay, 2023-11-17:
Hello,
Is it possible to prepare Home Assistant OS for use with Raspberry Pi Imager on PINN?

I'm currently using a CM4 on a Waveshare board with PINN installed on the eMMC, and I'd like to run the system from an SSD.

I've gone through the documentation, but it seems like I might be missing some JSON files required for this process. Is it feasible to accomplish this setup?

Alternatively, what other options exist to install it on the SSD? Disassembling a lot of technology would be necessary, so I'd prefer to explore this option only in case of a failure.

procount, 2023-12-03:
> Is it possible to prepare Home Assistant OS for use with Raspberry Pi Imager on PINN?

Maybe, I have not looked into Home Assistant, but if it runs on Debian or similar, then there is a good chance to convert it.

> I've gone through the documentation, but it seems like I might be missing some JSON files required for this process. Is it feasible to accomplish this setup?

The JSON files are used to describe your OS and inform PINN how to install it. For a new OS, they will have to be created, but you can use those from a similar OS as a starting point.

lurch, 2023-12-19:
> I'm currently using a CM4 on a Waveshare board with PINN installed on the eMMC, and I'd like to run the system from an SSD.

I don't have a CM4 myself, but I believe your first step would be to write PINN to the SSD, and then change the EEPROM config to boot from SSD instead of EMMC?
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#nvme-ssd-boot