procount/pinn #762: ssh credentials

zweistein22, 2023-11-22:
Hello, 

I have a problem with ssh credentials:
followied :  https://github.com/procount/pinn/wiki/Remote-installation-of-PINN
to install pinn on a raspberry pi sd card.

I generated the userconf file as per instructions with a known username and password on the rpi running bookworm 32 bit.    
After unzipping pinn-lite.zip  I copied the userconf in the /boot  directory and rebooted the rpi.
ssh is active, but the credentials are not working.
What can this be?
Could there be a problem that I created the userconf with openssl 1.1 and pinn is using openssl 1.02 ?

What other options do I have instead of a brute force approach?

zweistein22, 2023-11-22:
ok, as per readme root is username and it works. userconf doesn't seem to be used

procount, 2023-12-03:
> I have a problem with ssh credentials:
followed : https://github.com/procount/pinn/wiki/Remote-installation-of-PINN
to install pinn on a raspberry pi sd card.

This is not the usual way to install PINN to an SD card. It is for a very specific case of installing PINN onto a remote system over the internet that is already running another OS e.g. Raspberry Pi Os, Essentailly, it _replaces_ the original OS with PINN so that multiple OSes can be booted in the future.

> I generated the userconf file as per instructions with a known username and password on the rpi running bookworm 32 bit.
After unzipping pinn-lite.zip I copied the userconf in the /boot directory and rebooted the rpi.
ssh is active, but the credentials are not working.

The userconf file is for creating a new user on the Raspberry Pi OS that is subsequently installed using PINN. PINN does not use it itself. PINN uses the well known username/password of root/raspberry, as you have discovered.