procount/pinn #765: Android Things

dismal002, 2023-12-09:
Will support for Android things be added https://dl.google.com/dl/androidthings/rpi3/devpreview/3/androidthings_rpi3_devpreview_3.zip

lurch, 2023-12-09:
I've never used Android Things myself, but https://en.wikipedia.org/wiki/Android_Things says "Android Things Dashboard shutdown began on January 5, 2021. After January 5, 2022, Android Things Dashboard has been shut down completely and all remaining data has been deleted." so perhaps there's not much point in PINN going to the effort of adding it?

P.S. See also https://killedbygoogle.com/ :wink:

dismal002, 2023-12-11:
It still works as the image itself does not depend on the cloud. It is basically just a version of Android for IOT devices and the dashboard is not required :)

lurch, 2023-12-11:
...but presumably Google aren't updating it any more, which means any security problems it has won't get fixed?
(But of course I have no idea if it has any security vulnerabilities or not)

dismal002, 2023-12-13:
It is for embeded devices and they do not need to connect to the internet it is not pinn's responsibility for security at all as you are using a 3rd party OS. However there are no major priv esc type vulnerbilities such as kingroot or towelroot

bmx666, 2024-10-11:
> Any update?

I found that DP4 is still accessible - https://dl.google.com/dl/androidthings/rpi3/devpreview/4/androidthings_rpi3_devpreview_4.zip