procount/pinn #767: PINN keeps restarting after OS selection

IBilly75, 2023-12-14:
With RPI 4 I have the latest PINN with following OSes installed:

- Batocera RPi4
- Recalbox7
- Libreelec
- Volumio
- OSMC PI4

Suddently when I select any of the last 3 OS system restarts after 1-2 seconds.

Any idea what may have gone wrong in PINN?

I guess this is not OS related issue, since it affects 3 OSs the same way.

Many thanks

procount, 2024-01-01:
Assuming your attached log files are correct, there are some inconsistencies between the actual partitions listed by fdisk in log.txt and the contents of installed_os.json. I don't know how this can have occurred. Have you mnually messed with the partitions or altered the installed_os.json file at all?

procount, 2024-02-20:
Closing as mo response

closed 2024-02-20 by procount