procount/pinn #770: [Pi5] Lock-ups

procount, 2023-12-28:
Model: Pi5 
PINN: v3.8.9
Occasionally and randomly, PINN will freeze, requiring a reboot.
This seems to happen just after the wifi has established and the IP address has been set.

Poopi, 2024-01-28:
100% reproducible when moving the USB mouse

procount, 2024-01-28:
Yes I had suspected something similar myself.

Poopi, 2024-01-30:
Do you have a beta branch somewhere to look at?

Poopi, 2024-02-23:
Was thinking about source code rather that binaries that I already checked...

procount, 2024-02-23:
I need to get it into a fit state first, then I'll consider pushing it up.
The odd thing is that apart from the removal of the alternate bootloading mechanism, the PINN source code is virtually the same as before. The 6.1 kernel is the same across all the models, but only the 64-bit version for the Pi5 exhibits this issue.
I think it's more of a kernel issue, unless it's the kernel that's highlighting a latent problem in the application 🤷‍♂️

Poopi, 2024-02-25:
I do think it is the latter. I do have an old Xiaomi keyboard that generates spurious messages in kernel. When PINN hangs I can simply remove the dongle and the boot continues....

procount, 2024-05-24:
When it locks up, it seems to be stuck here:
```
(gdb) backtrace
#0  0xf6cc81c8 in pthread_cond_wait () from /home/pi5/buildroot/output/host/usr/arm-buildroot-linux-uclibcgnueabi/sysroot/lib/libpthread.so.0
#1  0x015f9750 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) bt
#0  0xf6cc81c8 in pthread_cond_wait () from /home/pi5/buildroot/output/host/usr/arm-buildroot-linux-uclibcgnueabi/sysroot/lib/libpthread.so.0
#1  0x015f9750 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```
So, probably a thread or stack error.....?