procount/pinn #771: [Pi5] Slow initialisation of USB

procount, 2023-12-28:
Model: Pi2, Pi Zero 2
PINN: v3.8.9

USB keyboard and mouse take too long to be detected, making pressing the Shift key ineffective at boot up to enter the recovery menu.
A workaround is to set `forcetrigger` in cmdline.txt

lurch, 2023-12-29:
Says "Pi5" in the subject line, but "Pi2" in the text? :thinking:

procount, 2023-12-29:
Maybe my nomenclature is a bit ambiguous. By [Pi5] I mean "PINN v3.8.9 beta suitable for the Pi5", which is actually also applicable to all models.

procount, 2024-01-02:
@lurch any idea why usb init could be so slow?

lurch, 2024-01-02:
Not off the top of my head. And I'm still away from home on my festive break, so I currently don't have any Pis for testing with either.
Have you got any necessary USB / keyboard drivers compiled directly into the kernel, and not as loadable modules?

procount, 2024-01-02:
No modules. All built in like before. Should be the same drivers. Maybe specific to that kernel (7?) Or in combination with 6.1 version 🤷‍♂️ 
Needs more analysis...