procount/pinn #772: Removing project space to free up space

carlhinrichsen, 2023-12-31:
I am looking to remove project spaces to give space back to the partitions of installed OSes.
I have Raspbian, Batocera, Lakka and a few other OS installed via PINN, as well as 8 project spaces.
Well, in fact, I somehow managed to uninstall/remove/merge project space 1 in a way that gave space back to the Batocera partition (and possibly a few others). Batocera literally had 0 space left, and now it has 1.4 GB.
I now only have project spaces 2-8 showing as installed. Unfortunately I cannot for the life of me remember how I did this, and cannot seem to reproduce. I'd like to do this again with other project spaces to free up more space. Not clear from the documentation how to do this. Any recommendations?

procount, 2023-12-31:
I have avoided any deleting, moving or merging partition functions within PINN as it can get quite complicated.
PINN maintains a mapping of the partitions used by each OS so it can select the correct partition to boot. But also each OS maintains references to each of its partitions so that it can mount them. If the partitions are altered, some or all of these potentially need updating.

This is not a risk-free operation, so I would always recommend taking a backup of your drive first, or maybe copying it to another drive and working on the copy.

Option 1 would be to use PINN to backup each of your OSes you a USB drive, and then reinstall them without the project spaces using PINN to preferably another drive, or the same one, if you have already backed it up. (Please check any PINN backups are successful before wiping your original SD card as backups should be ok but are not guaranteed).

Option 2 would be to remove your PINN drive (SD card or whatever) and mount it on another Linux system where you can use a tool such as GParted to remove your ProjectSpace partitions and then resize your other partitions accordingly. After this, you would need to manually fix the installed_os.json file and fix up all the other partition references.

If you choose option 2, I suggest you post a copy of your current `installed_os.json` file (from the /settings partition - on /dev/mmcblk0p5 if using an SD card) here, along with the output of `blkid`, then I can potentially guide you on the next steps.

carlhinrichsen, 2024-01-02:
Thank you so much for your detailed answer @procount . I'll do that. Out of curiosity, any idea how it's possible that my project space 1 isn't listed anymore (only 2-8 are installed) and I got some space back from it for my previously full OS partitions?

procount, 2024-01-04:
> any idea how it's possible that my project space 1 isn't listed anymore (only 2-8 are installed) and I got some space back from it for my previously full OS partitions?

Maybe you used GParted to delete the projectSpace and then enlarged the Batocera partition, finally editing installed-os.json to match 🤷‍♂️

closed 2024-04-07 by procount