procount/pinn #773: [RPi5] PINN Will Not Load AOSP14 Recovery Enviornment

NeppityNepNep, 2024-01-05:
So, I recently found this project and followed the instructions for updating my eeprom and the beta version of PINN, and I loaded up a few OS's (Ubuntu, Raspbian, and AOSP 14). All loaded up just fine, but when I toggled the switch to reboot recovery for AOSP such that I can install Magisk, Widevine, and GAPPS, it just refuses to load anything and just loads a black screen. I tried reinstalling and the fix options present in PINN, but no dice. Just wondering if there is anything I can do to get TWRP to load using PINN, or if there is an issue preventing it from happening.
[image]
Here is a picture of what the boot partition looks like for the android OS.

procount, 2024-01-08:
Apologies, I hadn't tested this feature, assuming it was the same as AOSP13.
I'll investigate to see if anything additional needs to be done.

procount, 2024-01-08:
I just freshly installed AOSP14 on my RPI5 and I was able to boot into the TWRP recovery program just by following the instructions in Konsta's FAQ.
Could it be that yours has rebooted into TWRP, but the black screen is due to a video issue? You could try using a different monitor, or try installing AOSP14 to an SD card on its own to see if you get the same behaviour.

procount, 2024-01-09:
If you are installing AOSP to USB, you need to change the dtoverlay to android-usb as detailed in Konsta's FAQ.

NeppityNepNep, 2024-01-09:
Ok, I missed that part my apologies. So I did that on my USB standalone stick and it works as intended. But doing the same thing on my PINN config.txt causes it to bootloop like it did on my USB before I set up the config.txt as intended.

NeppityNepNep, 2024-01-09:
So, the standalone external USB is functional with no issues with the usb dtoverlay.

On my drive with PINN, I edited the config to include the usb dtoverlay. That resulted in a bootloop, Revert that change and reboot AOSP from PINN and it boots normally to AOSP. After that I went into settings, clicked the toggle to reboot to recovery, rebooted to attempt to load TWRP through PINN and I get no signal out. Sorry for the confusion, hopefully that makes sense.

procount, 2024-01-09:
So,  I need to check installing AOSP wuth PINN **on USB** and check twrp recovery.... ok

NeppityNepNep, 2024-01-09:
That is very weird. I have tried reinstalling PINN and just AOSP and trying to reboot to recovery with only the drive connected and I still get black screen.

I dont know if this makes any difference, but I am using a SanDisk Extreme External SSD instead of a USB drive for PINN and AOSP. I am also using the official Pi 5 PSU, so power I dont think is a concern.

NeppityNepNep, 2024-01-09:
Also as an aside, I do notice some discrepancies between my PINN android partition and my USB android partition. Where F: is PINN and G: is standalone. Not sure if that has any significance.
[image]

procount, 2024-01-09:
The differences are expected and are due to PINN modifying the *.img files to be suitable for multi-boot.
If you post them here I will take a look at them to check they have been modified correctly and there's nothing odd because of your set-up.

NeppityNepNep, 2024-01-09:
Sure thing. Here are all the img files from my drive.
[Images.zip](https://github.com/procount/pinn/files/13881057/Images.zip)

NeppityNepNep, 2024-01-09:
[Other Boot Files.zip](https://github.com/procount/pinn/files/13881082/Other.Boot.Files.zip)
And here are the other boot files. That plus the image files would make up the entire boot partition.

procount, 2024-01-11:
I had a quick look at your img files and they seem to be in order, so I', a bit stumped why your TWRP recovery will not display.

NeppityNepNep, 2024-01-12:
Hmmmmm, maybe I can send an image file of my entire drive in its current state. I was able to successfully create images using macrium reflect and clonezilla if either of those would work.

procount, 2024-01-12:
If you can create an .img file of your SD card (that I can flash with Rpi Imager or `dd`), I can try and find some time to try it.

NeppityNepNep, 2024-01-13:
https://drive.google.com/file/d/1OoEgAbGPS5o39Zv5mG0zSb39L3JBss8d/view?usp=sharing

Made this using DD after shrinking a few partitions. Hopefully it works.

procount, 2024-01-13:
I have downloaded and installed your img.
When I booted Android, it went straight to the TWRP recovery screen. I tried this on 2 different monitors and both displayed it correctly. Maybe you should try a different monitor or cable? It displayed in 1920x1080 resolution.

However, it had problems rebooting back out of twrp into android, complaining about a /data partition, so when it rebooted it went straight back into TWRP.
My workaround was to tell it to reboot into recovery (even though it was already in it), and then reboot into the System, which then worked ok. No problem rebooting back into TWRP though. I will investigate that as a separate issue.

naidu72, 2024-03-02:
i am facing the same kind of issue with LineageOS 21 Android TV (Android 14) on Raspberry PI4 it looping to TWRP recovery mode. even after disabling GPIO21 port on  config.txt recommended by konstakang.com  but no luck

procount, 2024-03-04:
I just tried this on a Pi400, but had no issues at all. I could reboot to recovery and then reboot back to system.

naidu72, 2024-03-07:
can you please share me your config.txt file

k2helix, 2024-04-30:
I was having the same error and changing the monitor (in my case tv) worked.