procount/pinn #776: Add download via proxy.

paolosezart, 2024-01-09:
Request to developers to add possibility to download list and operating systems via socks proxy. My internet provider blocks or very much limits the speed to some internet sites. 

Here is a photo of the download speed using PINN image of raspberry pi OS:
[image]

And here is my internet speed:
[image]

Here is a speed measurement via proxy:
[image]

The internet is fast enough. However, PINN is very slow. Thanks.

procount, 2024-01-10:
Many of the OSes are hosted on Sourceforge, since it is free for open-source and has many mirrors around the world. Unfortunately, some of the mirror sites can be slow.
I am yet to find another free site that has similar facilities but with faster speed. I am open to suggestions.
Meanwhile I am looking into adding a mirror selection facility within PINN so that faster mirrors can be selected.

paolosezart, 2024-01-22:
> Meanwhile I am looking into adding a mirror selection facility within PINN so that faster mirrors can be selected.

This is a great idea! Is it possible in this case to add a speed check for these mirrors? Or is this already too complicated a process?