procount/pinn #778: Libreelec 9.28

JanSykor, 2024-01-13:
Is it possible to add version Libreelec 9.28 or Libreelec 9.26 image to **Legacy** bookmark?
Reason is that for Pi2 and Pi3 for some people is this version better then last versions due to better optimisation for HEVC video decoding.
Thank you
Jan

procount, 2024-01-18:
I'm working with the LibreELEC maintainers to update their offerrings in PINN.

JanSykor, 2024-01-18:
If there will be some problem I have from the past downloaded from Pinn version 9.2.6, so I can save it somewhere.
Thanks
Jan

procount, 2024-01-18:
It's ok I have it.

procount, 2024-01-19:
LibreELEC-rpi2_926 has just been added.

closed 2024-01-19 by procount

JanSykor, 2024-01-20:
Dear @procount 
Thank you very much - it is working. I only expected position i Legacy tab, but Media is also OK
Only Info button for this is not working - isi it possible to add to the file this address?
https://libreelec.tv/2020/11/02/libreelec-leia-9-2-6/

Thanks 
Jan

procount, 2024-01-21:
info URL is now updated.