procount/pinn #779: When I want to install os,I encounter with a problem as followed : http://raw.githubusercontent.com/procount/pinn-os/master/os/os all.json not connected

WangXin1708, 2024-01-13:


WangXin1708, 2024-01-13:
[image]

procount, 2024-01-14:
It looks like you have a problem downloading the list of operating systems.
Clearly your RPi 3B is connected to your network and you are accessing it via VNC, but can your RPi access the internet?
If you click on Close, and click on Close when the wifi dialog appears, does the list of OSes eventually appear? In this case, try setting the networktimeout setting to 30 seconds or more to give it more time before it times out.
Can you download the file manually through your web browser?

procount, 2024-02-13:
Closing as no response. Feel free to reopen if needed.

closed 2024-02-20 by procount