procount/pinn #780: [PINN PI5 ] does not work with HDMi 1024 x 600

hstriepe, 2024-01-17:
I have a Roadon 10 in touchscreen that works well with standard setups - both Rapberry PiOS and Ubuntu.

The initial PINN setup does not work with it, and there is no display. Connecting a multi-format display works. 

I have modified config.txt to the Roadon screen definition but it is ignored on boot.

Definitions for this screen are:

# Disable compensation for displays with overscan
disable_overscan=1
# Specific to Roadom 7 or 10 in screen 1024x600
framebuffer_width=1024
framebuffer_height=600
hdmi_force_hotplug=1
hdmi_cvt=1024 600 60 3 0 0 0 0
hdmi_group=2
hdmi_mode=87

procount, 2024-01-18:
Try modifying cmdline.txt by adding the following text to the start of the first (and only) line:
`video=HDMI-A-1:1024x600@60D`
`A-1` is for the first HDMI socket.

procount, 2024-01-31:
If you boot your Pi5 with this display and no drives attached, do you get a diagnostics screen?
Does it's output include `EDID: OK`?

Do you have an older model RPI (pi4 or earlier) that you can test it with PINN 3.8.7?

hstriepe, 2024-01-31:
Booting without any output gives me the Pi 5 diagnostic screen. MultiBoot (RonR) and all operating systems tried so far work. Pulling my dedicated Pi4 (Home Assistant) for test will take a bit.

procount, 2024-01-31:
If you can use your Pi4, please try PINN v3.8.7 with it. With that your config.txt settings in your first post _should_ work.
You can try the PINN 3.8.9 beta with your Pi4 as well, but your config.txt settings m\y not work. If not, add the `video=HDMI-A-1:1024x600@60`  to your single line of text in cmdline.txt.

procount, 2024-01-31:
> Booting without any output gives me the Pi 5 diagnostic screen.

But did you see the `EDID ok` message on the diagnostics screen?

hstriepe, 2024-02-01:
It worked well using the Pi 4 and v3.8.7.

hstriepe, 2024-02-01:
It looks like it's ok. Hard to make out, it's very small.

procount, 2024-02-01:
OK this may help.

Take your SD card with PINN 3.8.9 beta on it.
Download https://sourceforge.net/projects/pinn/files/beta/pinn-lite.zip/download and unzip it.
From the contents of the zip file, copy `kernel8.img` and `pinn.rfs` onto the PINN recovery partition of your SD card to overwrite what is already there.

Now you need to edit `cmdline.txt` to add `video=HDMI-A-1:1024x600@60` to the one and only line of text in that file.

Also add this to `config.txt`: `dtoverlay=vc4-kms-v3d`.
Insert it on the line below `initramfs pinn.rfs`
(It doesn't really matter as long as it is not after the [pi4] section.)

Now try booting it in your Pi5 and let me know how you get on.

hstriepe, 2024-02-10:
I reimaged. It booted ok to another HDMI.
I made the change you suggested and tried to boot. It died with a kernel panic - UFS: UNable to mount root  fd on unknown block.
I will go back to the beginning and reimage and just change cmdline.txt.

hstriepe, 2024-02-10:
Ok. Got it working on the last attempt.

- Imaged a 256GB card with pinn-3-8-9-beta.img
- System has an NVMe. Set boot priority to SDCard
- Connected and external HDMI viewer
- Booted and let it do the initial configure
- Turned it off. Edited cmdline.txt on a Windows 11 VM. This would not work on macOS.
    - Appended video=HDMI-A-1:1024x600@60 to its single line as originally suggested.
- Booted system to the attached Roadon 1024 x 600 monitor and it worked.
Now configuring the various OSs I want to try.

Thank you. Not sure why the cmdline.txt edit did not work the first time. Bu I did not follow this process.

closed 2024-02-12 by procount