procount/pinn #781: Complete newb needing assistance...

metalchef1, 2024-01-20:
Hello. I appear to be mostly computer illiterate, pretty much brand new to github (so not well versed in the community or protocol here) and definitely over my head with what I am trying to accomplish it seems.

I have a raspberry pi 4 b.

Lots of stuff when I google multiboot but I'm confused and don't seem to be able to find exactly what I'm looking for/applies to me? I can find pieces but I can't figure out how to connect them. 

I want to multiboot my pi from an hdd and if the pi can't power it than an sdd...

I have a custom image of retropie that is very large and I also want to have an os... rapbian should be fine and then since I believe I can use kodi from within that...I guess I just need the 2? But I may want the option of more later as I learn more?? 

I can find how to etch the custom image of retropi and run that... but then I won't have an os.
I can intall an os but can't figure out how to get the custom retropie into that...ect

I'm sure this is easier than I'm seeing it to be but its currently all over my head.

The retro pie image is a .img file and its 475 gb. I see everyone talking about needing media to be formatted in fat32 or fat16 and every article i can find about using an external hdd doesn't say anything about formatting..

Sorry to ramble... any help you or anyone else could provide to me would be greatly appreciated.

My favorite Jeremy Irons quote kind of describes my level of ability here...
 "Please, speak as you might to a young child or a Golden Retriever. It wasn't brains that got me here, I can assure you that."

procount, 2024-01-20:
OK, just a couple of questions first:
What capacity of Hard disk drive or SSD are you thinking of using? (Just so I can guide you in the best way to use it).
What size of HDD are you thinking of? 2.5" USB drives tend to be ok, but 3.5" drives tend to need more power so would probably need their own power supply, or if USB powered, then they may need a powered USB hub between your Pi and the HDD.
Do you know what version of Retropie is in your .img file?

metalchef1, 2024-01-20:
I would like to use this... As it's older and it's just sitting around..
https://a.co/d/fwt5aHm

If that won't work... I can use this..
https://a.co/d/6ZAvpdM

And the retro pi .img is..
Wolfenoz 512 gb recalbox image that I found on arcade junkies and downloaded from mega.

metalchef1, 2024-01-20:
And sorry... It's recalbox... Not retropi

procount, 2024-01-21:
I looked at the HDD, but I couldn't find any power requirement specifications for it there. Maybe you can google for more information on it. If the max power is <1A,then it should be ok, otherwise you'll need a separate USB powered hub. But if you have it sitting around, you may as well try it to see if it is ok on its own. I couldn't see any power specs for the SSD either, but generally they should be ok. Both of them should work fine with PINN. PINN only uses MBR partitioned drives, which limits it to using drives <=2TB.

Let me start with some brief background explanations for you.
All storage drives: SD cards, HDDs or SSDs etc are treated in the same way, despite the differing technologies that they use. Consider them as just a load of storage. Typically this is grouped into SECTORS of 512 bytes, but it needs some structure to it in order to store data on it. This is where FILE SYSTEMS come in. There are different types of file systems that use different techniques to store files with names such as FAT, NTFS, EXT, BTRFS etc. File systems are just a way of using and organising the sectors to store files on them. FORMATTING a drive is a way of putting a blank file system with its structures on a DRIVE.

As drives developed, they got so big that they became bigger than the filesystems could cope with. So the answer was to PARTITION the drive into several "virtual drives" also known as "partitions" or "volumes". Nowadays, the partitions are a way of just organising your drives for differing uses or differing Operating Systems.

The first type of partitioning was called MBR (Master Boot Record) which limits the size of the disk to 2TB. It can have a maximum of 4 primary partitions, but one of them can be converted to an extended partition which can contain many logical partitions. A more modern partitioning type is called GPT partitioning which is required for drives >2TB and is more flexible.

So, a new drive is first typically PARTITIONED into separate partitions, and each PARTITION is FORMATTED with a particular file system type which can then have files stored on it.

Image files (.img) are a low level bit copy of the disk You can take an image of a single partition , or of the whole disk, which will then contain several partitions. Image files typically require special programs to write them to a drive (e.g. dd, Raspberry Pi Imaging Utility, Etcher, etc). Image files contain the formatting structure AND the file data, so it is not necessary to format a drive if you are going to write an image file to it - writing the image will provide the file system format and the file data at the same time.

Most Linux based operating systems consist of 2 partitions. The first is a BOOT partition, which must be in FAT format as the boot loader only understands FAT. This will contain the kernel which understands how mount and access the files on the other partitions e.g. containing the ROOT file system. The bootloader will typically load the first FAT partition that contains certain bootfiles. To have multiple Operating systems on the same drive you need a multi-boot program to select between them.

PINN is a multi-boot installer that allows you to install, backup and restore multiple operating systems to the same drive, with a 
menu to select which one you want to run at boot-up. It manages the partitioning and formatting of the drive and relocates the operating systems to different partitions. It automatically shares out all of the drive space between the installed operating systems. Because it separates the partitioning and formatting from the installing of the files, it does not use .img files, but requires a separate .tar.xz archive file for each partition. PINN does provide installation files for Raspios, retropie and recalbox, amongst others. Although it cannot load your recalbox.img file directly, you can install a fresh recalbox installation in PINN and then overwrite it with the files from your custom.img by loop-mounting it and copying the files afterwards. You may need some more instructions on how to do this part, later.

As I mentioned above, PINN will share out all of our drive space between the OSes you first install, so there is no space to install other OSes later. However, there is a facility to install "ProjectSpaces" which are essentially empty OSes to reserve drive space so that they can be replaced by another OS (of the same partition layout) at a later date.

Hopefully that has given you a good overview of disk storage and multi-booting with PINN in simple terms.

If it sounds too complicated, you may also want to take a look at multi-boot from RonR in the RPI forums. It is mostly restricted to raspios-like OSes, but I believe it can install .img files directly.

metalchef1, 2024-01-21:
So... Use pinn (am I right in thinking you are the architect of pinn? From what I have seen online?) and put the 3 or 17 os I want on the drive...
Then open the drive in windows and copy the custom img file over the raspbian img file that was put during the pin install?

I've used pinn several times messing around mounting and testing os.. And that thing probably couldn't be more user friendly...

You make this sound like it's super easy?

I'll try to mess with it tomorrow if I'm able..

And thank you for the help so far and the lesson.. Greatly appreciated..

procount, 2024-01-21:
PINN is a fork (modification) of the original NOOBS program developed by several people at Raspberry Pi. So I didn't create the architecture of it, but more adapted and extended it into what PINN is today.

> Then open the drive in windows and copy the custom img file over the raspbian img file that was put during the pin install?

I don't think you can do this from Windows. You could use the RASPIOS image that you will also store on your target drive, or even use the recovery shell from PINN to do this.

As well as your target drive (HDD) you will need an additional drive that you can store your large custom .img file on (or flash to) and then mount this so you can copy the files across to your target drive under linux. (Windows does not understand the ext4 file system). Remember that an img file contains the filesystem structure (incl partitions) as well as the file data. As PINN has already created the partitions and the file structure you can't just (easily) copy an img file over the top. It's safer just to copy the files into the existing file structure.

You would overwrite the RECALBOX installation, not the raspbian/raspios installation (since your img is a recalbox image). I assume that was just a typo?
You just need to make sure that your fresh recalbox installation is big enough to hold your custom img.

metalchef1, 2024-01-21:
Right on... Very cool.

I'm going to have to read this a few times to get my head around it but I think I'm getting the picture??
Not sure..
But I'll try to mess around tomorrow if I can and see what happens..

metalchef1, 2024-01-22:
Good afternoon. Been missing around with pinn all morning and checking out different os and such to see what I want... Seems raspbian is the best os?
Libreelec seems fine except I can't get plex to run..
And batocera is going to be dependent on the custom build I downloaded.

So I'm just going to go with those 3 and if I want to change things later I'll just redo the environment.

So.. I have the custom img on the ssd and I want to use the hdd for the storage for the pi.

Do I need to format the hdd in fat32 before proceeding with the pinn install?

Did I need the sdd to have been formatted into fat32 before putting the img on it so it can be read by the pi?

procount, 2024-01-22:
> Been messing around with pinn all morning and checking out different os and such to see what I want.

So you have PINN installed already - what did you install it on (SSD, HDD, SD card, or...) and how did you install it (flashing a .img file, using RPi Imager, unzipping pinn-lite.zip etc)?

Sorry, but I'm a bit confused by your terminology of what you want to do, so some more questions for you I'm afraid....

>  I have the custom img on the ssd

Do you mean you have the img FILE on the SSD or have you INSTALLED the img file to the SSD so it is bootable?

>  I want to use the hdd for the storage for the pi.

What type of storage are you talking about here? 

> Do I need to format the hdd in fat32 before proceeding with the pinn install?

It depends on how you want to install it, but generally not these days. PINN is available as a zip file that can be just copied to your target device, in which case it must be formatted as FAT 32 first. But this is not practical when your HDD is 1TB. It would be better to use the Rpi Imaging Utility and install PINN as an img from there to your HDD. You will find PINN under the "Misc Utility" category.

This suggests to me that you want to put PINN on the HDD. So the other 3 OSes (raspios, recalbox and LibreELEC) will also be on that same drive. As your HDD is 1TB, and PINN divides the disk up equally between the OSes you install, this means that it will allocate roughly 333MB per OS, which will not be big enough for your 475GB custom image to overwrite the recalbox image. However, we can work around this.

> Did I need the sdd to have been formatted into fat32 ....?

All drives need to have a file system on them before they can be used. This can be done by formatting it or by flashing an img file (which if you remember will install the file system and the file contents at the same time). 

> ...before putting the img on it so it can be read by the pi?

OK now I'm confused again. Do you want to put your custom recalbox image on the SSD and PINN and the other OSes on the HDD? This is not so easy to do. You need to put ALL the OSes (or at least their boot partitions) along with PINN on the same drive.  (The RPi will only allow OSes to be booted from the boot drive that PINN is on). It may be possible to move some of the other partitions (root or storage partitions) onto the SSD. So when you say you want to use the SSD for storage, were you thinking of the storage partition of recalbox where your games are stored?

metalchef1, 2024-01-22:
I was just messing round using the pinn website and a micro sd card to test different os and see what i wanted.  

I simply have the custom recalbox img sitting on my 2tb ssd because my internal on my pc doesn't have enough room on it for the file.

I wanted to use the hdd for everything...all os on the one drive...but now it seems my hdd is corrupt so I am going to have to look into purchasing another ssd.

I just want 3 or 4 os to be on 1 ssd as a multiboot.
1 of those os would be the custom recalbox img.....that is currently parked on my ssd....

By storage I simply meant the device/drive that will always be connected to the pi.
That (what will now have to be a new ssd.... will be the "hard drive" of the pi unit.

I had thought you said earlier I would need the drive with the operating systems on 1 and then the custom recalbox on another so I would then copy the custom image over the non custom install that i would get when setting up the multiboot?

procount, 2024-01-22:
It's ok - I was just making sure so we were on the same page 😄. That all seems good.

> Can I use the pinn website so I can choose the partition sizes myself, download and put those on the sdd # 1 and then plug in sdd # 2 with the large custom recalbox image) to copy it over the one pinn installs?

Yes, that's one of the things I was going to suggest. Probably the best idea if you are already used to using it.
It sounds like you have most of it sorted.

The only complicated bit will be copying your custom image over the recalbox installation. Let me know when you get to that point and I can lead you through it.

metalchef1, 2024-01-22:
Okay. Cool. Thanks.
That's why I was messing around it with...
Just wanted to get used to doing the installs.

So with the sdd I can just copy the download file from the website over to it?
Does that add need to be formatted to anything special before copying over the file and plugging it into the pie?

Believe it's currently in exfat..

procount, 2024-01-22:
Exfat should be fine.

metalchef1, 2024-01-22:
Okay. Thank you very much.

Just have to wait for Amazon to lose my first shipment with the new ssd and then order a replacement for the lost one..

metalchef1, 2024-01-24:
Got my 1tb crucial ssd.

Usable memory was 931 gb so I did the pinn image for 930.

Chose 5 os and got an error message... [image]

[image]

So I just installed the 3 that were good to go.

I can always go back and redo this... Since I'm trying to learn I don't care if I have to repeat steps multiple times. Just curious what that error meant? 

I used an SD and raspberry pi imager to load the program for usb booting.
Copied the pinn image to the ssd.. Plugged it in and nothing...

So I used the imager to format it... Did it again... And booted right away.
After all of that... I think it might have been the cable I was using.

I made the partition for recalbox much larger than necessary. I believe the image is around 475gb..and I made the partition like 750gb..

What is the next step in this process??

procount, 2024-01-25:
That's not an error message - it's a warning.
There is only ever 1 SD card slot on a Pi, so it is always designated /dev/mmcblk0.
However there could be multiple USB drives, but the designation of /dev/sda, /dev/sdb etc is not fixed. They could change on each boot. Therefore the use of their device designation is not preferred. Instead PINN tries to use PARTUUIDs for USB drives. However, Android does not support PARTUUIDs, so the only way to make it boot on USB is to use the /dev/sda designation. But to make this work, you should only have 1 USB drive fitted to avoid the risk of the drive designations being swapped over and it not booting. Hence the warning message.

> I used an SD and raspberry pi imager to load the program for usb booting.
Copied the pinn image to the ssd.. Plugged it in and nothing...

The new SSD was probably formatted as exfat. It needs to be FAT formatted to be bootable.

As for the next steps, I believe you have copied your custom .img file to your original SSD. You will need to loop mount this image and copy the files over to your recalbox partitions. I will write up some instructions. Meanwhile, please post the contents of the installed_os.json file. If you boot the Raspios OS, then you will find this file on /media/(username)/settings.

metalchef1, 2024-01-25:
I am not finding that file..
Under my user name  I have 4 folders.
Overlay
Recalbox
Storage
System

Was not able to locate in what was not access denied.

And I ran a search on the filename and didn't seem to come up with anything?

metalchef1, 2024-01-25:
When I press ctrl + alt + f2.... Nothing happens...?

metalchef1, 2024-01-25:
Okay.
Got the shell.
Went in.

First code sent it to the settings folder.

Typed the 3 other commands... Upon hitting enter it just went to the next line (didn't indicate anything happened).
Back in raspbian.... Can't find the file?

procount, 2024-01-26:
It will be on the RECOVERY partition.
The easiest thing to do is to plug your drive into a Windows computer and you will find it on the first drive that mounts (called RECOVERY)

metalchef1, 2024-01-26:
got it..

[
 {
  "bootable" : true,
  "description" : "A port of Debian with desktop and recommended applications",
  "folder" : "/settings/os/raspios_arm64_full",
  "group" : "General",
  "icon" : "/settings/os/raspios_arm64_full/icon.png",
  "name" : "raspios_arm64_full",
  "partitions" : [
"PARTUUID=a96f6bb0-06",
"PARTUUID=a96f6bb0-07"
  ],
  "password" : "raspberry",
  "release_date" : "2023-12-05",
  "supported_models" : [
"Pi Model",
"Pi 2",
"Pi Zero",
"Pi 3",
"Pi Compute Module 3",
"Pi 4",
"Pi 5"
  ],
  "supports_backup" : true,
  "url" : "http://www.raspbian.org/",
  "username" : "pi"
 },
 {
  "bootable" : true,
  "description" : "LibreELEC is a fast and user-friendly Kodi Entertainment Center distribution.",
  "folder" : "/settings/os/LibreELEC_RPi4",
  "group" : "Media",
  "icon" : "/settings/os/LibreELEC_RPi4/icon.png",
  "name" : "LibreELEC_RPi4",
  "partitions" : [
"PARTUUID=a96f6bb0-08",
"PARTUUID=a96f6bb0-09"
  ],
  "password" : "libreelec",
  "release_date" : "2024-01-11",
  "supported_models" : [
"Pi 4"
  ],
  "supports_backup" : true,
  "url" : "https://libreelec.tv/",
  "username" : "root"
 },
 {
  "bootable" : true,
  "description" : "The retro-gaming OS! An all-in-one, plug-n-play retro-gaming console, supporting 100+ gaming systems!",
  "folder" : "/settings/os/Recalbox_-_Pi4",
  "group" : "",
  "icon" : "/settings/os/Recalbox_-_Pi4/icon.png",
  "name" : "Recalbox - Pi4",
  "partitions" : [
"PARTUUID=a96f6bb0-0a",
"PARTUUID=a96f6bb0-0b",
"PARTUUID=a96f6bb0-0c"
  ],
  "password" : "recalboxroot",
  "release_date" : "2022-01-30",
  "supports_backup" : true,
  "url" : "",
  "username" : "root"
 }
]

procount, 2024-01-26:
So we now know recalbix is stored on 3 partitions 10,11 &12 (0a, 0b, 0c). Next we need to check if those partitions are big enough.
I suggest you boot into Raspios, go to terminal and type 
`sudo fdisk -l /dev/sda`
And paste the output here.
Then attach your other drive with your custom image on it.
Find out where the drive is mounted e.g. /media/pi/drivename.
Open a terminal at that location and type
`sudo fdisk -l filename.img`
Replacing the filename appropriately. Post the output, name of custom image and its full path.

metalchef1, 2024-01-26:
Hopefully did this correctly.
You have the patience of a saint.

Disk /dev/sda: 931.51 GiB, 1000204886016 bytes, 1953525168 sectors
Disk model: CT1000X9SSD9    
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disklabel type: dos
Disk identifier: 0xa96f6bb0

Device     Boot      Start        End    Sectors   Size Id Type
/dev/sda1             8192     137215     129024    63M  e W95 FAT16 (LBA)
/dev/sda2           137216 1953505279 1953368064 931.4G  5 Extended
/dev/sda5           139264     204797      65534    32M 83 Linux
/dev/sda6           204800    1249279    1044480   510M  c W95 FAT32 (LBA)
/dev/sda7          1253376  277774333  276520958 131.9G 83 Linux
/dev/sda8        277774336  278822909    1048574   512M  c W95 FAT32 (LBA)
/dev/sda9        278822912  455565309  176742398  84.3G 83 Linux
/dev/sda10       455565312  461856765    6291454     3G  c W95 FAT32 (LBA)
/dev/sda11       461856768 1951408125 1489551358 710.3G  7 HPFS/NTFS/exFAT
/dev/sda12      1951408128 1953505279    2097152     1G 83 Linux

/media/metalchef1/Elements/Wolfanoz_TotalRecal_Rpi4_512GB.img

Disk /media/metalchef1/Elements/Wolfanoz_TotalRecal_Rpi4_512GB.img: 476 GiB, 511099421184 bytes, 998241057 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x5f169346

Device                                                         Boot     Start       End   Sectors   Size Id Type
/media/metalchef1/Elements/Wolfanoz_TotalRecal_Rpi4_512GB.img1           2048   6303743   6301696     3G  c W95 
/media/metalchef1/Elements/Wolfanoz_TotalRecal_Rpi4_512GB.img2        6305792 998045696 991739905 472.9G  7 HPFS
/media/metalchef1/Elements/Wolfanoz_TotalRecal_Rpi4_512GB.img3      998047744 998241056    193313  94.4M 83 Linu

procount, 2024-01-27:
There's a couple of ways to mount your image. Let's see if this way works first.
Boot into Raspios and attach your Elements drive as above.
try:
`sudo kpartx -l /media/metalchef1/Elements/Wolfanoz_TotalRecal_Rpi4_512GB.img`  
This should list the available partitions inside the image, similar to the above fdisk command.

It should display something like this (although this is from another image):
```
loop16p1 : 0 262144 /dev/loop16 2048
loop16p2 : 0 4194304 /dev/loop16 264192
loop16p3 : 0 524288 /dev/loop16 4458496
loop deleted : /dev/loop16
```
If you get any errors at any point, then we'll have to try another way.
Take note of the loop device names in the leftmost column. You will need these soon, but they will be different for you.

Now we repeat this command with different options to perform the loop-mount:
`sudo kpartx -ar /media/metalchef1/Elements/Wolfanoz_TotalRecal_Rpi4_512GB.img`  
This should have produced 3 new device mappings. 
To check, type
`ls /dev/mapper`
It should display something like:
`control  loop16p1  loop16p2  loop16p3`
With the same loop names as above.

(More to follow.....)

metalchef1, 2024-01-27:
doesnt seem to work for me

metalchef1@raspberrypi:~ $ sudo kpartx -l
sudo: kpartx: command not found

procount, 2024-01-27:
I thought it was part of the standard Raspios installation.
You can install it with:
`sudo apt install multipath-tools`
and try again

metalchef1, 2024-01-27:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  rtimucli
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  kpartx libsgutils2-1.46-2 liburcu8 sg3-utils sg3-utils-udev
Suggested packages:
  multipath-tools-boot
The following NEW packages will be installed:
  kpartx libsgutils2-1.46-2 liburcu8 multipath-tools sg3-utils sg3-utils-udev
0 upgraded, 6 newly installed, 0 to remove and 9 not upgraded.
Need to get 1,356 kB of archives.
After this operation, 9,667 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://deb.debian.org/debian bookworm/main arm64 kpartx arm64 0.9.4-3+deb12u1 [26.0 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 libsgutils2-1.46-2 arm64 1.46-3 [110 kB]
Get:3 http://deb.debian.org/debian bookworm/main arm64 sg3-utils arm64 1.46-3 [810 kB]
Get:4 http://deb.debian.org/debian bookworm/main arm64 sg3-utils-udev all 1.46-3 [37.0 kB]
Get:5 http://deb.debian.org/debian bookworm/main arm64 liburcu8 arm64 0.13.2-1 [69.5 kB]
Get:6 http://deb.debian.org/debian bookworm/main arm64 multipath-tools arm64 0.9.4-3+deb12u1 [304 kB]
Fetched 1,356 kB in 0s (3,731 kB/s)  
Selecting previously unselected package kpartx.
(Reading database ... 220777 files and directories currently installed.)
Preparing to unpack .../0-kpartx_0.9.4-3+deb12u1_arm64.deb ...
Unpacking kpartx (0.9.4-3+deb12u1) ...
Selecting previously unselected package libsgutils2-1.46-2:arm64.
Preparing to unpack .../1-libsgutils2-1.46-2_1.46-3_arm64.deb ...
Unpacking libsgutils2-1.46-2:arm64 (1.46-3) ...
Selecting previously unselected package sg3-utils.
Preparing to unpack .../2-sg3-utils_1.46-3_arm64.deb ...
Unpacking sg3-utils (1.46-3) ...
Selecting previously unselected package sg3-utils-udev.
Preparing to unpack .../3-sg3-utils-udev_1.46-3_all.deb ...
Unpacking sg3-utils-udev (1.46-3) ...
Selecting previously unselected package liburcu8:arm64.
Preparing to unpack .../4-liburcu8_0.13.2-1_arm64.deb ...
Unpacking liburcu8:arm64 (0.13.2-1) ...
Selecting previously unselected package multipath-tools.
Preparing to unpack .../5-multipath-tools_0.9.4-3+deb12u1_arm64.deb ...
Unpacking multipath-tools (0.9.4-3+deb12u1) ...
Setting up libsgutils2-1.46-2:arm64 (1.46-3) ...
Setting up liburcu8:arm64 (0.13.2-1) ...
Setting up kpartx (0.9.4-3+deb12u1) ...
Setting up sg3-utils (1.46-3) ...
Setting up sg3-utils-udev (1.46-3) ...
update-initramfs: deferring update (trigger activated)
Setting up multipath-tools (0.9.4-3+deb12u1) ...
Created symlink /etc/systemd/system/multipath-tools.service → /lib/systemd/syst
em/multipathd.service.
Created symlink /etc/systemd/system/sysinit.target.wants/multipathd.service → /
lib/systemd/system/multipathd.service.
Created symlink /etc/systemd/system/sockets.target.wants/multipathd.socket → /l
ib/systemd/system/multipathd.socket.
Could not execute systemctl:  at /usr/bin/deb-systemd-invoke line 145.
Processing triggers for libc-bin (2.36-9+rpt2+deb12u3) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for initramfs-tools (0.142) ...
update-initramfs: Generating /boot/initrd.img-6.1.0-rpi7-rpi-v8
update-initramfs: Generating /boot/initrd.img-6.1.0-rpi7-rpi-2712

multipath-tools v0.9.4 (12/19, 2022)
Usage:
  kpartx [-a|-d|-u|-l] [-r] [-p] [-f] [-g] [-s|-n] [-v] wholedisk
	-a add partition devmappings
	-r devmappings will be readonly
	-d del partition devmappings
	-u update partition devmappings
	-l list partitions devmappings that would be added by -a
	-p set device name-partition number delimiter
	-g force GUID partition table (GPT)
	-f force devmap create
	-v verbose
	-n nosync mode. Return before the partitions are created
	-s sync mode (Default). Don't return until the partitions are created
metalchef1@raspberrypi:/media/metalchef1/Elements $ sudo kpartx -ar /media/metalchef1/Elements/Wolfanoz_TotalRecal_Rpi4_512GB.img
metalchef1@raspberrypi:/media/metalchef1/Elements $ ls /dev/mapper
control  loop8p1  loop8p2  loop8p3
metalchef1@raspberrypi:/media/metalchef1/Elements $

procount, 2024-01-27:
Yay you got there.
Ok  now..
```
cd /tmp
mkdir 1 2 3
sudo mount /dev/mapper/loop8p1 1
sudo mount /dev/mapper/loop8p2 2
sudo mount /dev/mapper/loop8p3 3
```

metalchef1, 2024-01-27:
once again.. thank you for the patience

metalchef1@raspberrypi:/media/metalchef1/Elements $ cd /tmp
metalchef1@raspberrypi:/tmp $ mkdir 1 2 3
metalchef1@raspberrypi:/tmp $ sudo mount /dev/mapper/loop8p1 1
mount: /tmp/1: WARNING: source write-protected, mounted read-only.
metalchef1@raspberrypi:/tmp $ sudo mount /dev/mapper/loop8p2 2
mount: /tmp/2: WARNING: source write-protected, mounted read-only.
metalchef1@raspberrypi:/tmp $ sudo mount /dev/mapper/loop8p3 3
mount: /tmp/3: WARNING: source write-protected, mounted read-only.

procount, 2024-01-27:
Great, so far so good.
Next step is to delete all the recalbox files you installed with PINN, to just leave the partition structure behind.
As you are deleting things, it's important we get this right and you don't delete the wrong things.
So what is the output of 
`mount`
?

metalchef1, 2024-01-27:
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
udev on /dev type devtmpfs (rw,nosuid,relatime,size=1668988k,nr_inodes=417247,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,noexec,relatime,size=388328k,mode=755)
/dev/sda7 on / type ext4 (rw,noatime,stripe=8191)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
bpf on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=29,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=12110)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
ramfs on /run/credentials/systemd-sysctl.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
ramfs on /run/credentials/systemd-sysusers.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
ramfs on /run/credentials/systemd-tmpfiles-setup-dev.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
/var/lib/snapd/snaps/bare_5.snap on /snap/bare/5 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide,x-gvfs-hide)
/var/lib/snapd/snaps/core22_1035.snap on /snap/core22/1035 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide,x-gvfs-hide)
/var/lib/snapd/snaps/gnome-42-2204_143.snap on /snap/gnome-42-2204/143 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide,x-gvfs-hide)
/var/lib/snapd/snaps/core_16204.snap on /snap/core/16204 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide,x-gvfs-hide)
/var/lib/snapd/snaps/gtk-common-themes_1535.snap on /snap/gtk-common-themes/1535 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide,x-gvfs-hide)
/var/lib/snapd/snaps/hello-world_29.snap on /snap/hello-world/29 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide,x-gvfs-hide)
/var/lib/snapd/snaps/snap-store_963.snap on /snap/snap-store/963 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide,x-gvfs-hide)
/var/lib/snapd/snaps/plexmediaserver_427.snap on /snap/plexmediaserver/427 type squashfs (ro,nodev,relatime,errors=continue,x-gdu.hide,x-gvfs-hide)
/dev/sda6 on /boot/firmware type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
ramfs on /run/credentials/systemd-tmpfiles-setup.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,nosuid,nodev,noexec,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=388324k,nr_inodes=97081,mode=700,uid=1000,gid=1000)
tmpfs on /run/snapd/ns type tmpfs (rw,nosuid,nodev,noexec,relatime,size=388328k,mode=755)
nsfs on /run/snapd/ns/plexmediaserver.mnt type nsfs (rw)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
portal on /run/user/1000/doc type fuse.portal (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
/dev/sda9 on /media/metalchef1/Storage type ext4 (rw,nosuid,nodev,relatime,errors=remount-ro,stripe=8191,uhelper=udisks2)
/dev/sda8 on /media/metalchef1/System type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/sda12 on /media/metalchef1/OVERLAY type ext4 (rw,nosuid,nodev,relatime,errors=remount-ro,stripe=8191,uhelper=udisks2)
/dev/sda10 on /media/metalchef1/RECALBOX type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
/dev/sdb1 on /media/metalchef1/Elements type exfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,iocharset=utf8,errors=remount-ro,uhelper=udisks2)
/dev/mapper/loop8p1 on /tmp/1 type vfat (ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/mapper/loop8p2 on /tmp/2 type exfat (ro,relatime,fmask=0022,dmask=0022,iocharset=utf8,errors=remount-ro)
/dev/mapper/loop8p3 on /tmp/3 type ext4 (ro,relatime)

procount, 2024-01-28:
OK, to delete the default recalbox, you need to do the following:
(Be very careful that you are in the correct folder before deleting the files, else it could delete the wrong files)

```
cd /media/metalchef1/RECALBOX
sudo rm -rf *
cd /media/metalchef1/OVERLAY
sudo rm -rf *
```

Partition 11 is not yet formatted, so we need to do this next.
```
mkfs.exfat /dev/sda11
mkdir /tmp/SHARE
sudo mount /dev/sda11 /tmp/SHARE
```

procount, 2024-01-28:
Now we can copy the files from your custom image:
```
sudo cp -r /tmp/1/* /media/metalchef1/RECALBOX
sudo cp -r /tmp/2/* /tmp/SHARE
sudo cp -r /tmp/3/* /media/metalchef1/OVERLAY
```

procount, 2024-01-28:
Finally, you need to boot into PINN, holding SHIFT if necessary to enter the recovery menu.
Go to the Maintenance Menu.
Check the box next to Recalbox.
Select FIX from the menu.
select the "Re-run partition script" option and click ok.

Hopefully, that's everything done and you should now be able to boot into Recalbox.

metalchef1, 2024-01-28:
metalchef1@raspberrypi:~ $ cd /media/metalchef1/RECALBOX
metalchef1@raspberrypi:/media/metalchef1/RECALBOX $ sudo rm -rf *
metalchef1@raspberrypi:/media/metalchef1/RECALBOX $ cd /media/metalchef1/OVERLAY
metalchef1@raspberrypi:/media/metalchef1/OVERLAY $ sudo rm -rf *
metalchef1@raspberrypi:/media/metalchef1/OVERLAY $ mkfs.exfat /dev/sda11
exfatprogs version : 1.2.0
open failed : /dev/sda11, Permission denied

exFAT format fail!
metalchef1@raspberrypi:/media/metalchef1/OVERLAY $

procount, 2024-01-28:
Sorry add sudo.
`sudo mkfs.exfat /dev/sda11`

metalchef1, 2024-01-28:
just waiting on the second one to return my prompt...im assuming this wont be quick as the image is almost 500gb

metalchef1, 2024-01-28:
thought i would check in... should it be taking this long? still no prompt returned yet after 
sudo cp -r /tmp/2/* /tmp/SHARE

the first one was abit slow...but this a few hours now.

green light inside the pi is flashing slowly and regularly

procount, 2024-01-28:
It may take a long time....

It you want to monitor progress, why not open a second terminal and type `df -h`. It will show you how much space is used/available on each of your partitions, so you can monitor /tmp/share.

metalchef1, 2024-01-28:
/dev/mapper/loop8p1  3.0G  894M  2.2G  30% /tmp/1
/dev/mapper/loop8p2  473G  424G   50G  90% /tmp/2
/dev/mapper/loop8p3   60M   56M     0 100% /tmp/3

looks like its moving along

metalchef1, 2024-01-28:
[image]

procount, 2024-01-28:
Within PINN,
[ctrl]-[alt]-F2 again to get to the command shell.
login with root/raspberry
what is the output of
`ls /settings/os` ?

metalchef1, 2024-01-28:
Libreelec_rpi4 recalbox_-_pi4 raspios_arm64_full

procount, 2024-01-28:
and 
` ls /settings/os/recalbox_-_pi4` ?

metalchef1, 2024-01-28:
No such file or directory....

procount, 2024-01-28:
Check your typing. You posted above that it was one of the 3 folders in /settings/os.

metalchef1, 2024-01-28:
[image]

Am I missing something??

metalchef1, 2024-01-28:
Sorry... Didn't know it was case sensitive.

Icon.png os.json partitions.json slides (in blue) slides_vga (in blue)

procount, 2024-01-28:
Hmm, so why is partition_setup.sh missing from that folder? (rhetorical question!)
Let me send you a new copy to put in that folder.

metalchef1, 2024-01-28:
If it wasn't difficult it wouldn't be me.. Never met a piece of tech or anything with a current flowing through it that liked me...

procount, 2024-01-28:
Lol. :)

Oh! Recalbox is strange as it doesn't have a partition_setup.sh script for the default installation.
I'm going to have to dig into this a bit more....

(Each OS can be quite different and it takes a while to work out how to adapt them for PINN/multiboot. I'd forgotten Recalbox was odd)

metalchef1, 2024-01-28:
Naturally that's the one I have...

metalchef1, 2024-01-28:
Just a thought??
Could I use raspberry pi etcher to mount this custom image on the ssd by itself...
And do the other os with pin on an SD card?

I would have to unplg the ssd when I want to use the desktop but I feel I saw an option in etcher to use a custom img file?

procount, 2024-01-28:
You _could_....
But you're almost there.

procount, 2024-01-28:
I'm not sure how recalbox references its additional partitions, so I need to investigate.
But you could just try booting recalbox from PINN now and see if it works....?

metalchef1, 2024-01-28:
Okay.
I'll give it a shot.

metalchef1, 2024-01-28:
It loaded right up...
Lots to check out in it...

Thank you so much for the help.

There is no way I would have ever come close to figuring out most of what had to be done to accomplish that.

I "understand" what you did... But i don't understand how you did it...

procount, 2024-01-28:
Glad it worked out for you.

metalchef1, 2024-01-28:
Think I've got just 1 more question for you for now?
If I go into the os and the terminal and update all...
Will updates screw up this image? Since it's 2 yrs old and such?
Do I need to be cautious of what I do and how I do it so nothing gets messed up?

procount, 2024-01-28:
You can update most OSes in the usual way.
In any case, if it doesn't work, If you keep your custom image, you can always restore it as we did here.

[2 more replies]

closed 2024-01-29 by procount