procount/pinn #782: Cannot boot into Volumio after OS update

Hiforever, 2024-01-26:
I'm currently using Volumio relaese 3.546.  Once I update to OS to latest version within Volumio, Pinn fail to boot into Volumio again be it release 3.569 or 3.601,  

Update within Volumio OS works perfectly in the past.

procount, 2024-01-26:
Wihtout looking into this in detail, please try this -> https://github.com/procount/pinn/blob/master/README_PINN.md#re-run-partition_setupsh to see if it works.

Hiforever, 2024-01-26:
tried running the fixes includiing partition_setup.sh.   Still not working

procount, 2024-02-23:
Thanks for providing a solutuon. It does seem to work.
I have asked Matt to increase the default size of the boot partition to prevent this from happening on new installationz.

closed 2024-02-23 by Hiforever

arlo-infinity, 2024-02-23:
> Thanks for providing a solutuon. It does seem to work. I have asked Matt to increase the default size of the boot partition to prevent this from happening on new installationz.

No problem.  I realized what the issue was a few weeks ago once I found this the thread [Upgrade to v3.569 not possible] (including cases without PINN) (https://community.volumio.com/t/update-to-v-3-569-not-possible-all-you-need-to-know/63663), but I just had time to post it yesterday. Matt was super fast and has already updated the partition size to 120MB and upgraded the Volumio version to 3.616, so at least this boot partition issue should be ok for a little while.