procount/pinn #784: Error creating partition table - with M.2 and pi5 - pinn beta

JochenEb, 2024-01-31:
Hello,

ich have this
[https://www.amazon.de/ORICO-Upgraded-Ex ... B0B296B7ZH](https://www.amazon.de/ORICO-Upgraded-Externes-Aluminium-Heatsink/dp/B0B296B7ZH)

When I connect this case to the pi5 (Raspbian), it works without any problems and can write/read data. But if I now want to boot pinn-lite from it, it works, but when creating partitions pinn-lite aborts with an error.
"Error creating partition table."

Here again two debug files.
Debug01 = after the start of the pinn-lite
Debug02 = after the installation attempt by "Project Space - 1"

[pack.zip](https://github.com/procount/pinn/files/14108955/pack.zip)

procount, 2024-01-31:
Hi Jochen,
Thanks for copying your info to this github issue.
PINN uses the standard sfdisk tool to repartition your drive, but for some reason it is reporting that /dev/sda is not present, which it clearly is from the previous logs.
From the debug logs, it is not clear why or when the drive disappears. Is it before or after the sfdisk command has tried to repartition the drive? What is the state of the drive after the failure - has it been repartitioned?
After partitioning it should look like the table here -> https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained#update-for-noobs-v15-and-later, although partitions 6 onwards will not exist yet.

I have heard reports that this version of the kernel can be a bit slow to re-read the partition table so it may be just a timing issue. I think we will need to do some investigation, if you are willing?

Sorry, I have not even asked you about your Linux knowledge. Are you proficient at Linux, or able to get around under instruction? Do you have a spare SD card? Do you have any other RPis apart from your Pi5?

JochenEb, 2024-01-31:
The LW will not partition.
- MBR 223.57 GB - total
- 1. Partition "Recovery" 127 MB (fat16 - primary)
- 2. Partition "Settings" 32 MB (ext4 - logical)
- Rest (unallocated) 223.4 GB (logical)
Ich can send a photo.

I'm a few days older, my English skills are not good, I work with a translator.
My knowledge of Linux is limited, but with instructions I can definitely get by.
I have PI3, PI4 and PI5 here.
There are several free SD cards available.

JochenEb, 2024-01-31:
Still forgotten.
With a pi 4, pinn works without any problems, the LW can be partitioned without errors.
Attached is the LW's picture of PI5.
[image]

JochenEb, 2024-01-31:
Now I have completed the LW from the PI5, as it was, to a PI4 and selected a "Project Space - 1".
This worked without any problems.

Pinn V3.8.9a
[image]

procount, 2024-01-31:
I'm glad you have already tested your LW(?)* with PINN on a Pi4 - that was one of my first tests to get you to do. 😄 
Did you run this test with PINN 3.8.9 beta or the proper release of 3.8.7?
'* (what does LW mean? I assume it is your SSD enclosure)

OK, so let's concentrate on the PI5 and test the hypothesis that it is a timing issue.
I suggest you install PINN 3.8.9 beta to one of your SD cards and boot it in your Pi5.
Once booted, you can connect your LW which should appear as /dev/sda and will be mounted on /tmp/media/sda1 by PINN.

In PINN, you need to hold [Ctrl]+[Alt] keys down, then press and release Function key F2 and finally release [Ctrl]+[alt] to open a command shell. Login with `root` and password `raspberry`. 

This state is where we will need to do much of the testing from, so be familiar with how to get to this point.

I need to prepare some further instructions for you to do from here. Please bear with me as I only do this in my (very little) spare time. Meanwhile, can you post the output of `fdisk -l /dev/sda` so I know the current state of your LW?

By the way, if you are familiar with using SSH using PUTTY on a Windows computer, you may find it easier to login to PINN using PUTTY or another SSH client from your laptop. Then you can easily capture (cut and paste) any output that I may ask for.

JochenEb, 2024-01-31:
Ich work with Pinn v3.8.9a
Yes, LW (german: Laufwerk) is the uSB-SSD

fdisk -l /dev/sda
[image]
[image]

JochenEb, 2024-01-31:
I found my problem.
It was the original USB cable, which came with the case.
The cable works without problems on the PC, e.g. writing the pinn image to the SSD.
The same cable causes problems on the pi5, but it works on the pi4.
I don't understand it, but I can actually reproduce it now.

Sorry for troubling you with this problem.

procount, 2024-01-31:
How odd, but I'm glad you've solved your problem 😄

closed 2024-01-31 by procount