procount/pinn #787: OS OctoPi, Repetier-Server and OpenMediaVault

Bourns, 2024-02-11:
good evening everyone.

I tried to create pinn images for OctoPi, Repetier-Server and OpenMediaVault.

Unfortunately, this doesn't work and they are not yet available online. Have you ever done this?

I read the instructions and at least tried to work on them. But unfortunately I can't do it.

procount, 2024-02-11:
I've not tried any of these.
Last time I looked, OpenMediaVault is now only available as a script to be installed on top of raspios-lite. So your best bet is to install Rapios-lite and then install OMV in it. You can use PINN to take a personal backup of that, but it will contain your personal credentials, so not a good idea to share.
Are you sure the other 2 are available for the Pi5? Do you have  a link to their Pi5 download images?

Bourns, 2024-02-11:
Hi

thanks for the quick reply. Great tip for OpenMediaVault, thank you.

Yes, both OS are available for the Raspberry Pi 5.

Repetier-Server for 3D-Printer: https://www.repetier-server.de/download-images/
NightlyBuild OctoPi for 3D-Printer at the bottom: https://unofficialpi.org/Distros/OctoPi/nightly-arm64/ 
direct Link: https://unofficialpi.org/Distros/OctoPi/nightly-arm64/2024-02-11_2023-12-11-octopi-bookworm-arm64-lite-1.1.0.zip

JanSykor, 2024-03-01:
Hello
I am just checking the Octopi status and I think that it is still not ready for Pinn. There is still a few problems and Octopi can not be updated after installation like a Pinn - only Octoprint can be updated.
When it will be ready I will try to prepare Octopi for Pinn.
Jan

SMHRambo, 2026-01-03:
I wanted to know what you meant in this context (PINN). I know how to install octoPi.
OctoPi itself can be updated with a new image, but OctoPrint can be updated via the internal update process and the OS can be updated via apt.

procount, 2026-01-04:
@JanSykor 
1. Linux is case sensitive. In os.json, you called the os "Octopi" but you named the icon file "OctoPi.png" which doesn't match in case.
2. The json parser is not very forgiving and doesn't provide any information about any errors you may have. It's a good idea to validate your json files in https://jsonlint.com/ 
3. You left some trailing commas in the json files. Do not add a comma after the last item in a list.
4. the URL in os.json is malformed.
5. You changed the mount points in partition_setup.sh sed scripts from "root" to "rootfs" and "boot" to "bootfs". These are not partition labels. 
6. You have probably edited all the files with a DOS or Window editor, so the line endings are incorrect. You can change them to Linux line endings using dos2unix.
7. One of the slides is duplicated.
Other than that - a good effort!

SMHRambo, 2026-02-22:
After other projects and illness, I have finally managed to get back to work on OctoPi and PINN.

So, my image is running. After looking at @JanSykor's image, I noticed a few errors.
As procount already mentioned, there are some errors in the json file, and the naming was also a problem.

Link to the image:
https://mega.nz/file/PlR1yB5Z#Yqo2yOp-48Q-E9HReuB3lU5d8D_WyZ2TgclsrG3D8O0

I would appreciate your testing and feedback.

The following files needs to be changed.

All files in Unix LF.

os.json
```
{
 "description" : "A port of Raspberry Pi OS with the Octoprint",
 "group" : "Custom",
 "kernel" : "6.12",
 "name" : "octopi",
 "password" : "raspberry",
 "release_date" : "2025-12-29",
 "supported_models" : [
"Pi 3",
"Pi 4",
"Pi 5",
"Pi Compute Module 3",
"Pi Compute Module 4",
"Pi Compute Module 5"
 ],
 "supports_backup" : true,
 "url" : "https://github.com/guysoft/OctoPi?tab=readme-ov-file",
 "username" : "pi",
 "version" : "trixie"
}
```

partitions.json
```
{
 "partitions" : [
  {
   "filesystem_type" : "FAT",
   "label" : "bootfs",
   "mkfs_options" : "-F 32",
   "partition_size_nominal" : 256,
   "uncompressed_tarball_size" : 75,
   "want_maximised" : false
  },
  {
   "filesystem_type" : "ext4",
   "label" : "rootfs",
   "mkfs_options" : "-O ^huge_file",
   "partition_size_nominal" : 3905,
   "uncompressed_tarball_size" : 2854,
   "want_maximised" : true
  }
 ]
}
```

partition_setup.sh
```
#!/bin/sh
#supports_backup in PINN

set -ex

# shellcheck disable=SC2154
if [ -z "$part1" ] || [ -z "$part2" ]; then
  printf "Error: missing environment variable part1 or part2\n" 1>&2
  exit 1
fi

mkdir -p /tmp/1 /tmp/2

mount "$part1" /tmp/1
mount "$part2" /tmp/2

sed /tmp/1/cmdline.txt -i -e "s|root=[^ ]*|root=${part2}|"
sed /tmp/2/etc/fstab -i -e "s|^[^#].* / |${part2}  / |"
sed /tmp/2/etc/fstab -i -e "s|^[^#].* /boot\(.*\) |${part1} /boot\1 |"

# shellcheck disable=SC2154
if [ -z "$restore" ]; then
  if [ -f /mnt/ssh ]; then
    cp /mnt/ssh /tmp/1/
  fi

  if [ -f /mnt/ssh.txt ]; then
    cp /mnt/ssh.txt /tmp/1/
  fi

  if [ -f /settings/wpa_supplicant.conf ]; then
    cp /settings/wpa_supplicant.conf /tmp/1/
  fi

  if ! grep -q resize /proc/cmdline; then
    if ! grep -q splash /tmp/1/cmdline.txt; then
      sed -i "s| quiet||g" /tmp/1/cmdline.txt
    fi
    sed -i 's| init=/usr/lib/raspi-config/init_resize.sh||' /tmp/1/cmdline.txt
  else
    sed -i '1 s|.*|& sdhci.debug_quirks2=4|' /tmp/1/cmdline.txt
  fi
fi

umount /tmp/1
umount /tmp/2
```

SMHRambo, 2026-02-22:
Please note that the image used is partially outdated.
guysoft used an outdated version of his CustomPiOS, and the configuration of HAProxy and ssh used is not compatible with the versions of Trixi.
I'm trying to build a new version.

SMHRambo, 2026-02-25:
Here is my new OctoPi image, it is the latest version from the octoprint website.
https://mega.nz/file/6hIQlLpK#7xdvjPmxkfHniCDhaf3l_bQlIYbbFo62-8SUL_lrMEA

I wrote a bash script that automatically generates PINN images.
https://mega.nz/file/b8ZlxDqT#sGJCQXZf5dG_7e0mxv0aBJkoG8Xla46gJ-NojgcRTro

Simply unzip PINN.zip and use convert.sh.
`convert.sh -f imagefile -i iconfile`