procount/pinn #789: images in /os are not found on either SD or USB

Haas11, 2024-02-19:
Compressed the lineageos 21 ATV as .tar.xz and placed them inside the /os folder of both the sd card itself and an external usb. 
PINN V3.8.7d does not find either of them. 

unpacked the .zip file and re-compressed as .tar.xz for this. 
https://konstakang.com/devices/rpi4/LineageOS21-ATV/

did not try with other OS

procount, 2024-02-19:
You also need to provide some meta-data files to inform pinn about this OS.
See for example: https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN
Android is somewhat challenging to multiboot as some partitions must be provided as image files, and the partition_setup.sh script is quite complicated. You should review e.g. lineageos20 for the format and content of the files that are required if you want to do this conversion yourself, or you can wait until I publish this latest version.

Haas11, 2024-02-19:
Thank you for the swift reply.
I'm afraid my knowledge is not enough to fix this myself. 

I have settled for LineageOS20 for now who's repo is included with PINN (alongside Raspi64). 
Although now there is no network connectivity in LineageOS, neither through wifi nor through ethernet cable. 
A standalone LineageOS 20 runs without issue.

procount, 2024-02-19:
As you started with LineageOS21-ATV then switched to LineageOS20 - I wasn't sure if you meant the ATV version or not.
So I just installed both of them to my Pi400 through PINN and LineageOS20 had no problem connecting to ethernet or my wifi. Which is not surprising as is it the same software as running standalone. 
I'm not familiar with how to use LineageOS20-ATV, but it claimed to have connected to my wifi and I can ping it over the network. So I think you might have done something wrong.

procount, 2024-02-19:
LIneage OS 21 is now available

closed 2024-02-19 by procount