procount/pinn #791: [Pi5] Does not boot any of the installed operating systems

mkreisl, 2024-03-05:
Hello, I have here a Raspberry Pi 5 with 8GB Ram

The problem is that none of the operating systems (I have tested Ubuntu 64-bit and Pi-OS 64bit) that I have installed will boot.
Instead PINN reboots and offers me the boot menu again

If I insert the medium (I have tried it with a sd card and an external usb disk) into a Pi 4, the operating systems boot properly

Another test I did:
I have replaced my previous PINN sd-card files with the files from the ZIP file, here is the same problem. None of the operating systems want to boot in Pi 5, in Pi 4 everything is fine

What can that be?

procount, 2024-03-05:
It sounds like you need to update your firmware, as detailed in the announcement post
https://forums.raspberrypi.com/viewtopic.php?p=2173713#p2173713

mkreisl, 2024-03-05:
Yes, indeed. it was the firmware

I was actually sure it couldn't be that because I had already updated the firmware once and then lost track of it.
As I had seen earlier there was a new firmware shortly after I had done that.

Now it works great!

Thanks for the quick help

closed 2024-03-05 by mkreisl