procount/pinn #792: Errors installing LibreELEC

overkillcaraudio, 2024-03-09:
Hi, PINN is a great utility. I’ve been using it with not many problems. I’m new here. 
I’m getting this error with the LibreELEC OS. It doesn’t happen in noobs. It says to check the debug. 
it looks like something isn’t where it’s supposed to be. Some other guy wrote about this in the reviews so it’s not just me. Although I know that this it the right place to raise the issue.
Here’s a pic of the bottom of the dubug. 
Please check it out and let us know what you think.
Thanks for the product.
[image]

procount, 2024-03-09:
The file does not currently exist. I suspect it is because LibreELEC 12 is being released and I have some updating to do for them.
I expect it will be fixed shortly.

overkillcaraudio, 2024-03-10:
Cheers for the update. I’ll work out another way for now. I just wanted to check the problem wasn’t on my end. I’m teaching myself a lot from this, Thanks again for your work.

JanSykor, 2024-03-10:
Hello
I have same problem with Libreelec. I think Libreelec 12 not yet release, because Kodi 21 is just RC1
It can be problem on Libreelec site, or the image is saved on Pinn servers?
Where is saved Pinn debug log which Is mentioned in error message?

Thanks
Jan

procount, 2024-03-10:
Rpi4 System.tar.xz file has temporarily disappeared from the LibreELEC servers.
Their maintainer is not able to fix it for a week.
I've been able to put a temporary fix in for now for v 11.0.6

JanSykor, 2024-03-10:
It is working now
Thanks for your quick action
Jan
P.S. for the future use - where I can find installation debug log ?

procount, 2024-03-10:
Sorry, I meant to answer that, but forgot :(
It's in `/tmp/debug`
See also https://github.com/procount/pinn/wiki/Troubleshooting

overkillcaraudio, 2024-03-11:
Awesome, thanks again.

closed 2024-04-07 by procount