procount/pinn #796: MoOde Audio

neric34, 2024-03-17:
Hi,
I used PINN to install MoOde and LibreElec on a Pi4. It works very well. But I can't update MoOde... I think the new version isn't "in-place update", a new image is needed, I believe.
Could you update MoOde on PINN, please ?
I would like to try to do it but I don't have that much skills...
I'll maybe try with pinnify, if it's work ?

Thx

VortexNexus, 2024-07-27:
Hi,

@procount 
Using parts of the procedure defined in the [wiki](https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN), I tried to create my own partitions TAR files based on MoOde 9 (bookworm). But there was something strange with PINN installation of the OS. MoOde started normally and then after about 2-3 seconds restarted automatically and crashed. Then, I was not able to boot anything, having even PINN's installation broken.

As perhaps you know, during MoOde's boot up, a PHP script is launched that makes checks and configuration stuff. Among others, it checks the /boot/firmware/config.txt file, replacing it when it doesn't match the expected structure. Another important point is that MoOde uses PARTUUID to define its partitions in cmdline.txt and fstab files, so I had to update those files with the standard partitions paths instead of PARTUUIDs before creating the TAR files.

What I noticed while retrying, was that PINN defined MoOde's boot partition in fstab as **/dev/mmcblk0p01**, that means the very **PINN's boot partition** ... After that, the PHP script defines the config.txt file it founds (PINN's one) as broken and replaces it, breaking down PINN's boot process. When I manually update MoOde's fstab file before its first boot, then everything goes fine.

Could you tell me if PINN expects any special placeholders values for partition names in cmdline.txt and fstab files in the TAR files ?

@neric34 
It's quite an old thread,for sure, but the procedure defined in the [wiki](https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN) can help you. Be aware that it's a little old and some commands won't work without a little refresh through googling. But in the case you don't need more than one OS on the same PI, you can just install MoOde following its normal install process through Raspberry Pi Imager.

procount, 2024-07-27:
It looks like you are installing to SD card, so PINN will use the /dev/mmcblk0pX notation. If installing to USB, then it would use PARTUUIDs. I can't remember if you can force PARTUUIDs on an SD card. In any case, it is the responsibility of partition_setup.sh to set up the OS for multiboot by adjusting the cmdline.txt and /etc/fstab files. So the original values in the TAR files are irrelevant apart from pattern matching in the script file. You should be able to parse the parameters passed from PINN to the script and adjust the files as required.

VortexNexus, 2024-07-27:
Yes, I am installing on a SD card, you're right, using the /dev/mmcblk0pX notation. The partition_setup.sh script I use is the one present in the original [MoOde](https://sourceforge.net/projects/pinn/files/os/moode/) PINN install directory. A header comment says this is the standard Raspbian Full one. I don't understand why it doesn't work. It seems that the boot partition name given by PINN to the script's parameters is not the good one. This is perhaps due to my installation process: I asked PINN to create the data partition and two project reservations. Then I directly asked PINN to replace the first project by my custom MoOde image taken from my traditional os subdir ;-). What do think?

VortexNexus, 2024-07-27:
I tried another installation procedure asking PINN to create the data partition and directly install MoOde and one project reservation, but same result with MoOde automatically restarting and rewriting PINN's config.txt.
Could you tell from where is given the boot partition name to the partition_setup.sh script ? In the case the caller of this script is a system part, it could be modified in the MoOde system and therefor could be somehow "hardcoded" to the first partition of the device ...

procount, 2024-07-27:
Sorry, I have no time to look at the moment.
But if MoOde is based on raspios or the latest kernel, then perhaps its partition_setup.sh script should be looking in the /firmware folder for cmdline.txt......? Have a look at the raspios partition_setup.sh for similar changes.

I'm afraid there's no guarantee that things don't have to change when you try to convert an upgraded OS. It's at the whim of the maintainers. Sometimes it is easy, other times more tricky! 😄

VortexNexus, 2024-07-28:
So, my bad, you are right :sweat_smile:, sorry. The MoOde version in PINN's files is 8.0.2 based on RaspiOS Bullseye and kernel 5.15. From the 2023-10-10 release, /boot mount point has been moved to /boot/firmware in RaspiOS. And the partition_setup.sh changed, removing two space chars so that the sed command works ... So using the new version of RaspiOS' partition_setup.sh script solved my problem :smirk: 

As always, thanks a lot for your help and your work :+1:

dem23-tech, 2024-12-01:
Dears,
I am writting for some help here.
I am trying to install moode v.9. I have followed all steps listed here, but when I create the os in my USB and tried to load it in PINN, the install option is not activated.
I tried with other os from the os list from PINN and they work.
What I am doing wrong?

Thanks for your support!!

procount, 2024-12-01:
So can I assume you can see your new MoOde installation in PINN's list, but when you select it, the Install button isn't activated?
Have you updated the os.json, partitions.json and partition_setup.sh files appropriately?
Maybe post them here so I can see what you have?

dem23-tech, 2024-12-02:
Thanks for your quick reply.
Yes your assumption is right. 
Please, see my files:

os.json

{
    "description": "Moode 9",
    "name": "MoOde9",
    "password": "moodeaudio",
    "release_date": "2024-11-26",
    "supported_models": [
        "Pi Model",
        "Pi 2",
        "Pi Zero",
        "Pi 3",
        "Pi Compute Module 3",
        "Pi 4"
    ],
    "url": "http://www.raspbian.org/",
    "username": "pi",
    "supports_backup": true,
    "nominal_size":
    
}

partitions.json

{
    "partitions": [
        {
            "filesystem_type": "FAT",
            "label": "bootfs",
            "mkfs_options": "-F 32",
            "partition_size_nominal": 164,
            "uncompressed_tarball_size": 64,
            "want_maximised": false,
            
        },
        {
            "filesystem_type": "ext4",
            "label": "rootfs",
            "mkfs_options": "-O ^huge_file",
            "partition_size_nominal": 4284,
            "uncompressed_tarball_size": 3784,
            "want_maximised": true,
            
        }
    ]
}

partition_setup.sh

```
#!/bin/sh
#supports_backup in PINN

set -ex

# shellcheck disable=SC2154
if [ -z "$part1" ] || [ -z "$part2" ]; then
  printf "Error: missing environment variable part1 or part2\n" 1>&2
  exit 1
fi

mkdir -p /tmp/1 /tmp/2

mount "$part1" /tmp/1
mount "$part2" /tmp/2

sed /tmp/1/cmdline.txt -i -e "s|root=[^ ]*|root=${part2}|"
sed /tmp/2/etc/fstab -i -e "s|^[^#].* / |${part2}  / |"
sed /tmp/2/etc/fstab -i -e "s|^[^#].* /boot|${part1}  /boot|"

# shellcheck disable=SC2154
if [ -z "$restore" ]; then
  if [ -f /mnt/ssh ]; then
    cp /mnt/ssh /tmp/1/
  fi

  if [ -f /mnt/ssh.txt ]; then
    cp /mnt/ssh.txt /tmp/1/
  fi

  if [ -f /settings/wpa_supplicant.conf ]; then
    cp /settings/wpa_supplicant.conf /tmp/1/
  fi

  if ! grep -q resize /proc/cmdline; then
    if ! grep -q splash /tmp/1/cmdline.txt; then
      sed -i "s| quiet||g" /tmp/1/cmdline.txt
    fi
    sed -i 's| init=/usr/lib/raspi-config/init_resize.sh||' /tmp/1/cmdline.txt
  else
    sed -i '1 s|.*|& sdhci.debug_quirks2=4|' /tmp/1/cmdline.txt
  fi
fi

umount /tmp/1
umount /tmp/2

```

Thanks for you r help here

procount, 2024-12-03:
I have updated moOde to v9.
The installation slideshow seems a bit messed up, but it doesn't affect the installation.
Maybe you can find out what the differences are between your conversion and this one.

closed 2024-12-03 by procount

dem23-tech, 2024-12-03:
Thanks a lot!!

Your support was great!! I will try tomorrow to install it from scratch and try to find what was not working out.

Thanks again!

neric34, 2024-12-04:
Big thanks ! It works very well for me ! :)