procount/pinn #797: Rasberry Pi5 fails to boot into PINN

allanchbook, 2024-03-19:
I have tried several ways

1 Downloaded files from Browser implementation

2 Downloaded Zip File from Soundforge

Both same results
Cannot find OS to boot

I checked your EEprom Recovery Zip file
Seems for Pi4  Readme file
Add files dated December 2023

My EEprom File 5 January 2024

If you require more details Please contact me

procount, 2024-03-19:
This Pi5 version is still a beta, so it is not quite so easy to install, I'm afraid. It's still a bit too big for a upgrade from the usual version, but that should be happening soon which will make it easier for you.

> 1 Downloaded files from Browser implementation

Not sure what you mean by this

> 2 Downloaded Zip File from Soundforge _(it's actually sourceforge)_

ok, but it's how you install it to your drive that really matters. Probably some issue that that you haven't detailed. 

> My EEprom File 5 January 2024. 

It should be ok if you have updated it recently, that was mainly for first adopters before RPT had updated their files.
But that was to fix an issue with PINN not booting into an installed OS, whereas you don't seem to be able to boot into PINN to install any OSes in the first place, so it's a different matter.

What type of device are you installing to - an SD card? (I'll assume this for now).
If you are comfortable installing normal OSes to an SD card with Rpi-Imager, then I suggest you use that to install "Raspios arm64 lite". Plug it into your Pi and make sure it boots. Once you have confirmed that works, remove it and plug it into your PC again. The first BOOT partition of your SD card will be about 512MB in size and contain several *.dtb files and an overlays folder amongst other things. Make a mental note of how it looks, then delete all of those files and folders so that it is now blank (Make sure you are deleting these files/folders from the correct drive/partition!

Now you can unzip the contents of the pinn-lite.zip file (from the sourceforge beta folder) onto your now blank boot partition.
Check that it looks similar to what was there from Raspios before. Now place your SD card back into your Pi and turn it on. PINN should start and reformat your SD card appropriately.

allanchbook, 2024-03-20:
Used 128GB Flash Drive......NOT SD card

I did as you suggested

PINN l selected :-
    Android OS
    Raspberry Pie Full OS

No Wired Connection
Connected  via WiFi 
Both OS's downloaded
Android OS booted Fine
Raspberry Pi OS  failed to boot to graphical user interface - only to Busybox command prompt no login

is it possible to load partitions from SD card Working from booting from that SD card

I intend using Data partition as a home directory for multiple  OS's

procount, 2024-03-21:
> Raspberry Pi OS failed to boot to graphical user interface - only to Busybox command prompt no login

That is unusual. Were there any error messages? Maybe try re-installing Raspberry Pi OS.

> is it possible to load partitions from SD card Working from booting from that SD card

Maybe, although there is always a risk of copying a partially modified file. Better to do it from another OS, like PINN's recovery mode, then you know that there will be no open files. Also you you need to adapt some files to account for the partitions being on a new drive and partition, e.g. cmdline.txt and fstab. Re-running the partition_setup.sh script from the FIxup menu is sometimes suffcicient.

allanchbook, 2024-03-21:
Used 128GB Flash Drive......NOT SD card

I did as you suggested

PINN l selected :-
    Android OS
    Raspberry Pie Full OS

No Wired Connection
Connected  via WiFi 
Both OS's downloaded
Android OS booted Fine
Raspberry Pi OS  failed to boot to graphical user interface - only to Busybox command prompt no login

I also noticed when i press Shift Key to get into PINN menu it froze no response had to reboot 

is it possible to load partitions from SD card Working from booting from that SD card

I intend using Data partition as a home directory for multiple  OS's

procount, 2024-03-21:
Freezing is a known issue at the moment with pi5.

allanchbook, 2024-03-21:
Thanks for your help

Think i will leave it for now

Any ideas on best information to understand boot on Pi5 to see if i can implement it myself
Am quite used to creating boot menus on Linux
And creating Partitions using Gparted

Wish i could help you but have no idea where to start

First time trying Pi

procount, 2024-03-21:
Google is your best friend 😉 
Fortunately, the freezing is intermittent. Just pull the plug and power up again until it works. Try not to use any USB peripheral until the network has established - that tends to reduce the occurrences.

allanchbook, 2024-03-22:
Thanks for all help and advice

Very much appreciated

I am not desperate to get this done

Tried reading the official Documentation not much help ....infact more
confused

Think i will try again in a few months time

Regards
Allan

procount, 2024-04-07:
Closing for now. Please reopen if this recurs.

closed 2024-04-07 by procount