procount/pinn #798: Beta v3.8.9d Bug - kde2lee64 No Longer Boots

annaclets, 2024-03-21:
I have been using for my RPi5 the beta version 3.8.9 without any issues. Today, after updating to version 3.8.9d, kde2lee64 no longer boots, always ending up in the Busybox login.

I made another PINN SD card with version 3.8.9 and kde2lee64 boots successfully. Seems like there's a bug in version 3..8.9d.

By the way, is there a way in PINN to downgrade to a previous version?

procount, 2024-03-21:
Sorry, I screwed it up. Working on a new version now.

So, 3.8.9d prevented a previously installed kde2lee64 from booting? 
I can understand a new installation not booting, but not an existing one..

I have reverted the beta to the previous version. If you download and unzip pinn-lite.zip over your existing PINN, it should downgrade.

annaclets, 2024-03-22:
I installed version 3.8.9e today, and there are no issues whatsoever.

Thanks!