procount/pinn #799: Official 7" Touchscreen

panthersnc, 2024-03-21:
I tried installing PINN multiple times on my Raspberry Pi 5 with no success. I'm using the official 7" touchscreen which I believe the issue is PINN not connecting to the DSI 0. I have tired the img in the beta folder and I have also tried installing an os from raspberry pi imager and then wiping the boot partition and copying the new files over. I tried adding the 7inch dtoverlay also with no success. I am running the upgraded firmware. Current setup is Raspberry Pi 5 with the new active cooler and the official 7" touchscreen connected to DSI0

procount, 2024-03-21:
ThIs Pi5 version is still in beta because several things are not working or untested as yet on the PI5, the touchscreen being one of them. Also CEC and Pimoroni hyperpixel drivers are not tested, for example.
The Pi5 only works with KMS, so you will have to add the appropriate dtoverlay for the touchscreen and set any specific video modes in cmdline.txt. 
I am trying to address these in turn as I have time.

steveiliop56, 2024-06-03:
Is there a workaround to make it work?

steveiliop56, 2024-07-31:
@procount do you have any updates on this?

bullemer, 2025-06-29:
Any updates ? I am having the same issues.