procount/pinn #800: Not using language file during cold boot

JanSykor, 2024-03-26:
The boot menu in cold boot is allways in english.
Booting over recovery menu (after push SHIFT button) is correct in selected language, but when you do not push SHIFT the next boot menu is in english. Maybe same reason for this is why first message about pushing SHIFT is allways in english although you have select another language and this message is already translated.
Same behaviour in version 3.8.7.f and 3.8.9.d
Sorry I cannot find where is the problem in the code.
Not big problem, just information.
Thanks
Jan

procount, 2024-03-26:
Not something I normally test, so thanks, I'll look into it.

procount, 2024-04-07:
Fixed in 3.8.8. will be merged into PI5 later.

closed 2024-04-07 by procount