procount/pinn #802: Rpi5 update leads to small font

ZebMcKayhan, 2024-04-02:
Hello and thanks for your great work.
I have been on 3.8.9d on my rpi5 for some time now without issues. Updated a couple of days ago to 3.8.9f but then all fonts went crazy small. I can't even see what the text says even if I put my nose to the TV. 
I can press + a maximum of 3 times and I can start reading the text but just barely. But it won't let me increase size more. 
Windows size seems fine though.
I've tried setting various cmdline.txt resolutions and I can see it changes aspect and such but the font stays really small.
And at the next reboot font size adjustment with + have been forgotten.
Have something gone wrong or am I missing some setting?

procount, 2024-04-02:
The latest version now uses the KMS video drivers, which resulted in fonts that were a point or 2 larger, so I compensated by dropping the font size down similarly so the font size should be the same as before.
I see you have already tried using the '+' and '-' keys to adjust the font size. This is not remembered.
What resolution are you running at?
Check your config.txt file. It should have have the following lines at the end:
```
[board-type=0x17]
kernel=kernel8.img
overlay_prefix=overlays6/
dtoverlay=vc4-kms-v3d
```
What happens if you remove the `dtoverlay=vc4-kms-v3d` line ?

ZebMcKayhan, 2024-04-02:
I didn't specify resolution just now, but probably 1080x720 since that's my tv resolution. How could I tell which pinn uses currently?

Removing that line makes the fonts proper size. Altough I get white border on each side (both splash screen and in menu). Perhaps wallpaper_resize would fix that?

procount, 2024-04-02:
It sounds like KMS is not setting the right resolution.
Without KMS, you can use fbset or tvservice to find out the current resolution, then maybe try setting that on the cmdline when switching back to KMS.

I've changed the wallpaper to reduce the size of the installation, but the aspect ratio is not ideal. Use wallpaper_resize to fix it for now.

mkreisl, 2024-04-03:
I have the same problem. Ridiculously tiny font when I use the Raspberry Pi 5
On a Raspberry 4 everything looks normal, except for the creepy background with white stripes left and right
At the end if config.txt I see:
```
[board-type=0x17]
kernel=kernel8.img
overlay_prefix=overlays6/
dtoverlay=vc4-kms-v3d
```

procount, 2024-04-03:
I'm not seeing this ☹️
Let me do a fresh install later and I'll check.
Removing the vsc-kms-v3d overlay seems to fix it temporarily.

mkreisl, 2024-04-03:
> I'm not seeing this ☹️
> Let me do a fresh install later and I'll check.
> Removing the vsc-kms-v3d overlay seems to fix it temporarily.

Strange, but once again as usual: everything works for the developer :smile:
Deleting the line in question also brings the characters back into readable form for me

ZebMcKayhan, 2024-04-03:
I've opened the terminal within pinn and tried 
tvservice -s
But I only get 
`[E] Failed to initialize VCHI (ret=-1)`
I also tried other options but with same reply.

So I tested "fbset" which gives:
`mode "1920x1080"`
So I put in 
`video=HDMI-A-1:1920x1080M@60D`
In cmdline.txt as first in the line and added back the removed line in config.txt.
The result are similar as before, except I now get the white borders (which I didn't have before when using kms). Tiny unreadable fonts. Back to removing dtoverlay option again.

procount, 2024-04-03:
I should have mentioned, tvservice will not work on the Pi5 - you'll need to run it on another Pi model. The aim is just to see what the real resolution of your TV/Monitor is.

ZebMcKayhan, 2024-04-03:
Here you go, sorry for the bad quality, downsizing didn't treat this picture well:
[image]

procount, 2024-04-03:
This is what I see with all the default settings (i.e. using KMS) on my 1920x1080 HDMI monitor.
This is just a clip of a 1920x1080 screen. It has the white bars at the side of the screen, 
[image]

procount, 2024-04-03:
By contrast, this is what mine looks like when I comment out the KMS overlay
[image]

Here the font size is a couple of points smaller as PINN is compensating for the larger font in KMS that is not present, but it is still readable and easily fixed by one or two presses of the '+' key.

ZebMcKayhan, 2024-04-03:
I don't know what to say.

If I don't specify resolution in cmdline.txt and use KMS I don't get the white borders. 

If I don't use KMS I can push + probably 8 times to get really big fonts. But when using KMS it stops increasing the fonts after 3 pushes (for some reason i need to wait 1-2 seconds between pushes or it wont take). It's readable but just barely.

I don't know why it differs between us. Could we have different revisions of rpi5 boads? Or could it be the TV?

I don't have any issues with raspberry pi os which also uses KMS right?

I'm perfectly fine disabling KMS and I don't need perfectly sized fonts but I would like to contribute to this fine development wherever I can.

ZebMcKayhan, 2024-04-03:
Slightly better picture to compare with yours:
[image]

procount, 2024-04-03:
> I think the font is also a bit smaller on a raspberry pi 5 without the kms line than on a raspberry pi 4.

This is because PINN is assuming that KMS is in use on the Pi5 and is dropping the point size regardless.

procount, 2024-04-03:
> Slightly better picture to compare with yours:

That font is SOOOO tiny! But why?

ZebMcKayhan, 2024-04-03:
> I wondered if the font size was normal, but the resolution is crazily big....?
Adding
`video=HDMI-A-1:1920x1080M@60D`
To cmdline.txt have no/little effect on my font size. But I get the white borders so it seems to affect something.

mkreisl, 2024-04-03:
> A photo would do, (but you may have to shrink it), or maybe connect to PINN via VNC and do a screen capture.

BTW, VNC does not work on Raspberry Pi 5, at least with previous versions I used on Pi 5. Black screen gotten on TV and on VNC window. Raspberry Pi 4 is ok

procount, 2024-04-03:
Works for me 🤷‍♂️ My pictures above are from screenshots using VNC from my laptop.
(You did enable vncshare in the options?)

mkreisl, 2024-04-03:
> (You did enable vncshare in the options?)

Of course I did. Pi5 -> black screen, Pi4 working. With the Pi 5, just setting the option in the cmdline.txt meant that the picture no longer appeared on the tv

mkreisl, 2024-04-04:
It remains the same, neither with **vncshare** nor with **vncinstall** I get a picture in the VNC client. I have tried it with and without **dtoverlay=vc4-kms-v3d** in config.txt.

It seems to me that we are working with completely different PINN versions

procount, 2024-04-04:
I downloaded the 3.8.9f beta from sourceforge to be sure I was using the same version and not some development version. You can see the version and build date in my screenshots.

I really don't understand why vnc is not working fir you.

procount, 2024-04-04:
I'm working on the update procedure to include the PI5 beta version in the released version of PINN.
Once done, I'll put some debugging information in the logs to see if we can find out what is going on with fonts and vnc,

mkreisl, 2024-04-04:
> Once done, I'll put some debugging information in the logs to see if we can find out what is going on with fonts and vnc,

That should be helpful. I looked in **/tmp/messages** and couldn't find anything enlightening

procount, 2024-04-15:
I just released p3.8.9j beta. Hopefully the last beta before I merge it to mainline.
This just has a couple of debug messages in **/tmp/debug.**
The first, fairly early on, indicates the geometry of the screen that has been detected.
The second indicates the fontsize whenever you use +/- to change it.

( You can view the log in the GUI if you go to the maintenance menu, check the box next to PINN and select the fix up menu/ There's an option there to display the debug log which will pop up in a separate window. Rearrange/resize your windows to see them appropriately. It doesn't refresh, so click the refresh button to update the log.)

ZebMcKayhan, 2024-04-15:
> ( You can view the log in the GUI if you go to the maintenance menu, check the box next to PINN and select the fix up menu. There's an option there to display the debug log which will pop up in a [...]

Thanks! I've updated and the fonts stay the same. I've viewed the debug log but it's extremely hard to read, even the biggest fonts I can set.
Here are what I imagine to be relevant (from the top)
`error during vchi_initialize()
...
Geometry: QRect(0,0 1360x768)
...
Default display mode is "0"
...
Using fontsize 11
Using fontsize 12
....
Using fontsize 20 `

The first part after Geometry is fuzzy, I'm not 100% sure it's correct, it what it looked like to me.

I don't have a proper keyboard connected so I cannot get to ssh to save the log. Maybe I could try later this evening but it's difficult working with the tiny fonts.

A button when viewing the log to save it would be nice. Perhaps in the future?

procount, 2024-04-15:
Please try adding `video=HDMI-A-1:1366x768@60` to cmdline.txt

mkreisl, 2024-04-16:
I have started 3.8.9j briefly, fonts still look tiny as expected. To make it look normal, I have to use the '+' button to maximize the font size.
What remains, however, is the font in the header, which does not change.
By the way, the wallpaper just looks creepy, it takes me back to the early 70s. But that's just my personal opinion

procount, 2024-04-16:
I've not done anything with the fonts except output some debugging info about the fontsize and screen geometry. What values do you get in /tmp/debug?
If those values are as expected, I'm not sure why your fonts are coming out so tiny.

Sorry about the wallpaper, but the addition of the Pi5 kernel has really filled up the recovery partition so I am saving space where I can so everyone can upgrade to include the Pi5 support. A new PINN installation will use a 128MB partition instead of a 64MB partition. If the upgrade procedures detects this larger partition, it will revert to the old familiar wallpaper.

If you choose to add the runinstaller option in the config, then PINN will expand its partition size to 128MB on its next boot, but this will also repartition your drive and wipe out any installed OSes.

If you have space on your partition, you could always extract wallpaper.jpg from pinn-lite.zip and copy it across manually.

ZebMcKayhan, 2024-04-16:
> Please try adding `video=HDMI-A-1:1366x768@60` to cmdline.txt
> 
> > I don't have a proper keyboard connected so I cannot get to ssh to save the log.
> 
> You need the keyboard to get to the command shell, but SSH is a network protocol to access the command shell over the network from another computer. It's the best way to capture the log.

I added `video=HDMI-A-1:1366x768@60` to cmdline.txt but it does not change anything.

I clicked the SSH option and rebooted, but still could not access via SSH, I only get:
`Cannot negotiate, proposals do not match.
Key exchange was not finished, connection is closed`

Testing SSH I've noticed that most of the time the wifi is slow to make the connection so it times out and opens the wifi dialog. Without me doing anything the wifi connects by itself shortly after and I just hit cancel. After this SSH is never started. I had to reboot like 10 times before it made the connection without timing out. Then atleast port 22 is open...

procount, 2024-04-16:
> Testing SSH I've noticed that most of the time the wifi is slow to make the connection so it times out and opens the wifi dialog.

I usually add networktimeout=30 secs to avoid that.

ZebMcKayhan, 2024-04-16:
And here is mine:
/tmp/debug: https://paste.c-net.org/SplatHorses
/tmp/messages: https://paste.c-net.org/HooplaPharaoh

Hope you find anything useful in there.

procount, 2024-04-16:
> That doesn't work here. I always copy the content of pinn-lite.zip into an existing image which is 200MB in size
and I still have this shitty wallpaper.

Ah yes, sorry, a small mistake. I will correct that.

procount, 2024-04-16:
I've updated the pinn-lite.zip beta image.
It won't autoupdate because the version number is still the same, but you can use it to restore the proper wallpaper ;)

procount, 2024-04-17:
@mkreisl @ZebMcKayhan  - Please download [pinn.rfs](https://sourceforge.net/projects/pinn/files/beta/pinn.rfs/download) and copy it to your PINN recovery partition to see if it has any effect on your font issues. (It should present as version 3.8.9k)

mkreisl, 2024-04-17:
> @mkreisl @ZebMcKayhan - Please download [pinn.rfs](https://sourceforge.net/projects/pinn/files/beta/pinn.rfs/download) and copy it to your PINN recovery partition to see if it has any effect on your font issues. (It should present as version 3.8.9k)

The font is now larger everywhere except for the headline.
It is now the second smallest font I can set with +/- keys, one step bigger would probably be perfect, but as it is it is already much better :smile:

ZebMcKayhan, 2024-04-17:
> @mkreisl @ZebMcKayhan - Please download [pinn.rfs](https://sourceforge.net/projects/pinn/files/beta/pinn.rfs/download) and copy it to your PINN recovery partition to see if it has any effect on your font issues. (It should present as version 3.8.9k)

Yep, fonts look ok even without using the + key. The main window name font is still small and so is the initial message advising you to press shift key to get to recovery menu. But all else looks great! Nice work!

ZebMcKayhan, 2024-04-17:
Don't know if it is of any value to you but if I let the splash screen time out to get to the boot selection menu, the font is still small.
But if I go into recovery menu and then choose exit to the boot selection the fonts are normal.

procount, 2024-04-17:
That's good news. It seems to be something to do with how KMS interprets the DPI of your monitors. As I had specified a point size for the font, QT had scaled that by the DPI to get the number of pixels which became way too small. By specifying the font size in pixels it seems to have fixed it. I've just got to fix all the other little places you've identified.

procount, 2024-04-17:
@ZebMcKayhan - regarding SSH, Are you using the built-in wifi?
I've just discovered that the firmware changed (surprisingly some time ago!) to enable power saving mode by default, so although the wifi can connect to external websites to download stuff, it won't enable incoming connections from ssh or vnc. It works ok with a wired connection though.

ZebMcKayhan, 2024-04-18:
> @ZebMcKayhan - regarding SSH, Are you using the built-in wifi? I've just discovered that the firmware changed (surprisingly some time ago!) to enable power saving mode by default, so although the [...]

I am indeed using onboard wifi but when I got the networktimeout=30 so it makes the connection in time, SSH is working as intended and opens port 22 for incoming connections for me.
I've had some issues initially with the keys, but it just resolved itself after a number of reboots.

procount, 2024-04-19:
Your beta should shortly update to 3.8.9l
Please can you check the fonts are now legible when using KMS on PI5, for all your corner cases?
Thanks.

mkreisl, 2024-04-19:
The update was offered and executed, but something broke and a boot of it would no longer work.

I reinstalled and the font size looks fine to me now, I can make it 2x smaller, in the previous version it was only 1x. But the font in the title bar is still tiny for me.

Wallpaper is also fine.

ZebMcKayhan, 2024-04-19:
Uptade went fine for me.
Wallpaper font is normal size.
Recovery menu is normal size except title bar.
Boot selection menu is normal size except title bar.

Altough I still get the blue and white wallpaper despite I'm on rpi5 and had initial install from the beta and have 128Mb partition.

procount, 2024-04-19:
Thanks both for your testing.
I only wish I could reproduce these font issues myself so I could see which bits aren't working.  😉 
I'll have to do some more investigation on the titles.

procount, 2024-04-20:
Previously, I had specified the font size using a point size. This gets converted to pixels by scaling by the monitor's DPI setting. 
When using KMS, it seems a different value of DPI (dots per inch) density is reported, so very tiny fonts are produced. 
I have fixed this by specifying the font size in pixels to avoid this scaling.
However, it appears that the main window font is governed by the Window Manager directly and cannot be altered within QT. I also notice that some titles in the ui can only be specified by a point size. So this workaround doesn't work everywhere.

From what I have read, KMS gets this DPI setting from the EDID data from your monitor, probably the actual screen size parameter, so I think we need to understand what is different about your EDID data so it can hopefully be corrected somehow.

Can you please try and extract the EDID data from your respective monitors? Here is one article I found about it. https://blogs.gentoo.org/mgorny/2018/09/15/overriding-misreported-screen-dimensions-with-kms-backed-drivers/

Could you boot into Raspios and post the output of `xrandr`?
Or maybe `xrandr --verbose`?

ZebMcKayhan, 2024-04-20:
Sure,
Output from brands--verbose: https://paste.c-net.org/SucceedRoadside

I also extracted edid.bin if it's of any use to you: https://paste.c-net.org/ResumeHinting

mkreisl, 2024-04-21:
Here is my edid: https://paste.debian.net/hidden/0b7e5a88/

Generated by command `edid-decode  < /sys/class/drm/card1-HDMI-A-1/edid`

procount, 2024-04-24:
Thanks for the EDID files. 
I don't think I'll be able to solve this on my own (especially as it works ok for me), but to get expert help I need to create a small program that runs on proper bookworm that demonstrates the problem, so I'm going to focus on that. I will probably need you to test it for me to demonstrate the issue, if that's ok. It may be a little while. Meanwhile I believe removing the KMS overlay is a temporary workaround.

ZebMcKayhan, 2024-04-24:
Understood, no problems.

I thought that you would be able to use our EDID.bin on your system and this be able to see same problem as we? But perhaps I'm wrong: https://github.com/raspberrypi/linux/blob/rpi-5.9.y/Documentation/admin-guide/kernel-parameters.txt#L987

procount, 2024-04-24:
Yes, I hoped so too and I've already tried that but it wouldn't even boot with your EDID file 😞 
And I can't see anything wrong with the EDID file, but then I'm not an expert on that.

ZebMcKayhan, 2024-04-28:
Also related to KMS it seems I cannot get PINN to boot when tv set is off.
Previously I have always been adding 'hdmi_ignore_cec_init=1' and 'hdmi_force_hotplug=1' to prevent rpi to turn tv on during boot (as I woke up some nights that the tv was on probably due to power issues)
Now, these seem to prevent PINN from booting, it just hangs and when I turn on the tv it's just a black screen. Ssh is never turned on so I don't know which state PINN is in. Need to pull the power plug to get back.

Disabling KMS and I don't have this problem.

Or is there some other config parameter needed?

procount, 2024-04-28:
Try adding `vc4.force_hotplug=1` to cmdline.txt

ZebMcKayhan, 2024-04-28:
Yep, that fixed it. Thanks!