procount/pinn #803: 3 of 5 distros with no display on boot

rountree, 2024-04-04:
Hi all,

I'm a first-time PINN user and tested 5 distros on my new CanaKit RPi 4 Model B.  Kali64 and DietPi_64 worked as expected.  raspios_arm64, ubuntu 2310 and mxlinux mx23 appeared to install without error but on reboot never brought up video.

I'm using the Raspberry Pi Imager v1.8.5 to put PINN p3.8.7f on my sdcard.

First question:  Is this worth pursuing?  If the expectation is that distro support is hit-and-miss, no worries.

Second question:  If it's worth pursuing, what's the right venue?

Third question:  If it's worth pursuing here, what information do you need from me to see why some work and some don't?

Thanks.

procount, 2024-04-04:
Hi Barry and welcome!

From your profile you seem to be quite experienced, but as you are a first time PINN User with a _new_ Canakit, does that mean you are also a new Pi-User too? I ask because raspios_arm64 should be the most stable and reliable of the OSes. Have you tried installing raspios_arm64 (or any other OS) directly from the Rpi Imager without PINN?

Let me first address your questions:

1. Yes, this is worth pursuing. Converted distros should just work. The conversion process is just a format change from .img to .tar and the production of some meta files, so the result should be more or less identical to a standard install.

2. Here is a good venue to discuss.

3. Most likely this is not a PINN, raspios, or Pi4 problem, but something to do with your setup or the way the software has been installed. Let's discuss some possible common issues and how to troubleshoot these.

- What size SD card do you have?

Installing 5 distros will require quite a lot of space. The size indicated for each OS in PINN is the minimum size required, but you need to leave enough free space for the OS to work properly. If you have shoe-horned 5 distros onto an SD card that is only just big enough for them, then it may prevent them from initialising or updating.
Try installing each of your failed OSes with PINN one at a time and see if they work individually

- Install with RPi Imager

Try installing each failed OS individually using RPi Imager. This may work as a result of the OS having more disk space as above, but it would also eliminate PINN as being an issue.

- Video monitor/cable

Often when an OS does not show any video on boot, it does not mean it hasn't booted, but rather that there is just a problem displaying the video display.  The Pi is a little susceptible to certain video resolutions. The Pi4, for example has issues with certain resolutions around 1360x768. The introduction of using the KMS drivers has also introduced some issues when the tv/monitor does not have a compatible EDID output. 

If possible you could try swapping your display for another and/or changing the video cable.

Another way to check if the Pi has booted and it's just a display problem is to check it has a network connection.
Use a wired ethernet cable using DHCP. Check if the LED lights on the ethernet port flash after it has booted.
If they do, then you can check at your router if an IP address has been assigned. 

Hopefully, one of the above tests will resolve your issue. If not we can go further. There is also a troubleshooting section on my [wiki ](https://github.com/procount/pinn/wiki/Troubleshooting) which can guide you through creating some logs which might provide some more insight.

rountree, 2024-04-04:
@procount Thanks for the quick response.

As to my experience level:  last year I had an amazing group of interns write an Arm assembler in raw machine language.  We used RPIs until our Arm server got set up.  This year I have another intern who is writing an OS for RPIs, and a second intern who is doing the RPi Linux-from-Scratch experience.  I have enough experience to (usually) stay one step ahead.

The LFS project is what prompted me to retire my existing Model 3 with its consistent voltage droop issues and get a kit where I could trust the power supply.  LFS requires its own partition, and the stock Raspberry Pi Imager doesn't provide that capability.  NOOBS is defunct, of course, so I landed on PINN.

As to this issue:  I wasn't trying to install five OSes at once.  I was initially trying to install a single OS with the `provision` flag set in `recovery.cmdline` to give me some room to create the LFS partition, but that failed across all the distro installs I tried (some failures more spectacular than others).  I abandoned that and decided to just install two distros and later canabalize the partitions for the second, which led to another round of failures.  Finally, I decided to back off to installing single distros, one at a time, without any changes to the default.  That's what led to the list.

So no, I'm not trying to install five at a time....

I think it's less likely that is this is monitor/cable issue, but you're right, that's the next thing that needs to be checked.  The cable is fresh out of the box from Canakit, the monitor worked fine on my Model 3 and works fine (reproducibly) with the two distros I mentioned, so this feels like a distro-specific settings issue.  That said, I'm happy to take some time this evening to add a few more datapoints with a second cable and monitor and confirm via ethernet what's going on.

One last note:  I did eventually stumble over the "Utility" menu and was able to add the extra partition that way.  I didn't see mention of that anywhere in the docs.  Depending on how the summer goes, you might get an intern-led LFS FAQ PR.  That's a great feature, and I'd love for it to be more discoverable.

procount, 2024-04-04:
Just a quick response for now:
1. The `provision` setting was a request to provide some spare space so the SD card or SSD could do some wear levelling. I don't think it is utilised much. I don't use it.
2. Better to use a ProjectSpace. It is documented here -> https://github.com/procount/pinn/blob/master/README_PINN.md#project-spaces
3. You can create your own projectspace template if you need a different partition layout or size - they are just a bunch of Meta-files. You will need to change a setting to make it bootable from the menu. Ask me if you get to that point.
4. Please check you can install Raspios successfully with RPi Imager.

Raspios no longer has a default user account, so to test SSH access you will need to add a `userconf` file (see https://github.com/procount/pinn/wiki/Remote-installation-of-PINN#process) and a blank `ssh` file to the Raspios boot partition.

rountree, 2024-04-04:
@procount Go ahead and mark this as resolved.  As you suspected, it was the monitor.

In case someone is compiling a list.

Less reliable:
LG 22MP48HQ (2017)

Reliable so far:
BENQ XL2411 (2019)

Would it be a good use of time to figure out why some distros work on both and some only work on the BENQ?  Would it be useful for PINN users to get a warning that "While PINN appears to be working on this monitor, the greyed-out distributions use default settings that are likely outside the monitor's capabilities."

rountree, 2024-04-04:
You're also right that the project spaces documentation was right there all along.  I suspect I latched onto the idea that an empty partition would have to be specified on the command line, so I didn't look any further than that section.

Thanks for the fast, detailed responses.  I really appreciate it.

closed 2024-04-04 by rountree

procount, 2024-04-04:
The change to KMS drivers has caused a number of issues with monitors as it relies on accurate EDID information. There are various posts on the forum about this, but it is possible to override the EDID & video resolution so that the monitor still works. This is outside the scope of PINN, though really.

Glad you got it working. Feel free to ask for any further information you need.