procount/pinn #805: Update documentation regarding installation of PINN OS

mbernardi1961, 2024-04-05:
Since PINN OS has been added to Raspberry Pi Imager it would make sense that all the documentation reflected this, rather than suggesting use of various different formatting tools on differing operating systems.

procount, 2024-04-07:
It's mentioned on the 4th line of the first post of my sticky post in nice bold blue writing.
This is not the only way to install PINN and not appropriate for all use cases.
In LInux there are always at least 3 ways to do the same thing 😄. PIck whichever ways is suitable for you.

mbernardi1961, 2024-04-08:
True but the ReadMe.md on the front page of GitHub in the Quick Start section, doesn't mention it FIRST. Also NOOBS is no longer supported by Raspberry Pi (it's not on the downloads page anymore).

The quickest way to install PINN OS is to use the Raspberry Pi Imager, select Misc utility images, and select PINN.

No need for additional formatting tools and is available for Windows, Mac & Linux (including pre installed on Raspberry Pi OS).

TerranceBoyle, 2024-05-01:
Nothing shows up for Pi 5 

[image]

Am I missing something or does this only go up to Pi 4?

Many thanks

procount, 2024-05-01:
The version for the Pi5 is still in beta. There are still some missing features and niggles with it, but I hope to make it properly available soon as the main functionality is there. If you want to try the beta, see the link in the first post of my sticky thread on the forum, but be prepared to experience some issues with it.

procount, 2024-05-10:
Documentation partially updated

mbernardi1961, 2024-06-01:
> Documentation partially updated 

While README_PINN.md has added my suggestion, README.md has not and the latter is the first thing seen in the GitHub repo.