procount/pinn #807: New 3.8.9M beta

procount, 2024-04-26:
There's a new beta version about to drop 😃 
I'd appreciate some testing to find any bugs.

This version builds in the ability to change the partition sizes like on Matt's pinn.mjh.nz webpage for new installations.

There's no documentation yet, so just use it to install some OSes on a new SD card and see if you can work it out! Hopefully it's fairly intuitive.

It's still in beta, so expect bugs and problems.
The Pi5 version is still prone to locking up, for example, Only the GUI locks up, but it is necessary to pull the power as the mouse/keyboard is not working. If you have ssh access, you can still login and issue a rebootp command to recover.
Any questions, just ask.

JanSykor, 2024-05-04:
Hello @procount 
When I testing your new version on Pi3 I have problem with OSMC installation - there was some error messages with missing some noobs configs files and partition size tool not started - probably related to ability change of partition size - when I unselect OSMC and select Project space instead it was working.
Missing files was: os-pi2.json, partition_setup.sh, marketing.tar, partitions-pi2.json and NOOBS-Logo-2.png from http://download.osmc.ctv/installers/noobs/
I think it is not related, but on previous installation (not sure, maybe 3.8.9f?) on Pi5 I have back old Pinn background which I like it and in this version I have "cube" background.
In the tool "Adjust size of OSes" is possible to adjust only Extra (MB)? It could be possible to adjust also Nominal (MB) - it is gray now?
I experienced in the past, that some systems is not possible to install in Project space because low memory available in nominal section.

Thanks 
Jan

procount, 2024-05-04:
Maybe it was just a temporary internet glitch. I can download all of the OSMC versions without problem.

procount, 2024-05-04:
Project space partitions are now 512MB each

JanSykor, 2024-05-06:
> Maybe it was just a temporary internet glitch. I can download all of the OSMC versions without problem.

I can confirm, that OSMC installation and bigger Project space is now working.

Reagarding Nominal size it was misunderstanding - I thought that nominal means boot partition, not minimal total memory.
Now i understand what you meant.

Thanks 
Jan

JanSykor, 2024-05-06:
After update to 3.8.9o version I still have not old Pinn wallpaper background. It was already changed?
Thanks 
Jan

procount, 2024-05-06:
How big is your RECOVERY partition? 63MB or 127MB?

JanSykor, 2024-05-06:
I have 128 MB
There is 2 jpg file wallpaper and wallpaper2 - but both is same picture and size
[image]

procount, 2024-05-06:
On your PINN recovery partition, could you please try editing BUILD-DATA?
On the 2nd line, change 3.8.9o to 3.8.9 by just removing the 'o'.
This will force PINN to update itself again on next boot.

I have just updated the zip file, so provided the mirrors have updated, it should be corrected on next boot.
If it doesn't fix it, please wait a few minutes for the mirrors to update and repeat.

JanSykor, 2024-05-06:
Yes it helped
Now I have correct backgroud
Thank you so much. 
Jan

procount, 2024-05-06:
I'm hoping this is the last beta before merging the Pi5 into mainstream, so please give it a good testing, especially the partition resizing as it is new.
The Pi5 still has the same occasional gui lockup, no CEC, no DSI display, no Hyperpixel, no additional wifi etc, but I will try to address those after the merge.

andriilive, 2024-10-08:
@procount noticed joystic drivers, cool :D  should not **vl805** be included here also?

Raspberry Pi 4 EEPROM bootloader rescue image
*********************************************

The Raspberry Pi 4 contains a small EEPROM used to store the bootloader.

This rescue image reverts the bootloader EEPROM to factory default settings.

This rescue image also updates the USB 3 controller (VL805) firmware to the
latest version, 138a1, which has better full-speed isochronous endpoint 
support.

Raspberry Pi 4 board revisions 1.1 and 1.2 contain a separate EEPROM 
which contains firmware for the USB 3 controller (VL805): on newer revisions 
the USB controller firmware is stored in the bootloader EEPROM along with 
the bootloader.

[image]

procount, 2024-10-08:
Updating of any vl805 or other EEPROMS is best done with the already supplied RPF tools. I'm not sure I need to duplicate this.

andriilive, 2024-10-11:
> Updating of any vl805 or other EEPROMS is best done with the already supplied RPF tools. I'm not sure I need to duplicate this.

I'm noob in hardware, but this is USB 3.0 driver. My external drive turns off each boot when screen splashes, even being connected to a power cable

[image]
[image]

I guess a driver is a good start to start investigation, so as far I understand:

- PINN got updated so I can boot now with card card and drive inserted, so drive stucks only once (or it didn't so my sd with new nnnm / usb / sd bootliafer + os folder bypassed) (maybe a long time ago, brew don't auto update casks so I can't say)
- it should be a part of my boot
- it will be spread on projects install (and update, maybe?)

andriilive, 2024-10-11:
Here are some minor bugs i found:
the `pi-imager` generates `user-data` following it's global pattern, 

- but the **password** is probably random = unknown i guess, i can't login with root / raspberry or any expected passes i used in installations. The pass field is disabled while ssh key persisted in a config

[image]

- the keyboard layout: (**"us"** in my case) is being ignored, **"gb"** instead is being persisted and populated across instances,  **making me really angry on £** 

- a core network config is being ignored, not a bag but can be parsed / combined with network / dhcp configs

[image]

- same as above mentioned, dozens of errors in repo versioning, i just downloaded manually to sd card and installed from there (what is bad and not recommended because android symlinks or mounts images from it's locations, not recreates them from a scratch)

```bash
#!/bin/bash

# example: curl "https://sourceforge.net/projects/pinn/rss?path=/os/kali64" | grep "<link>.*</link>" | sed 's|<link>||;s|</link>||' | while read url; do url=`echo $url | sed 's|/download$||'`; wget $url ; done

# Array of OS names
array_of_os=("kali64" "raspios_armhf_full")

# Loop through each OS in the array
for os in "${array_of_os[@]}"; do
    # Create a directory for the OS and navigate into it
    mkdir "$os" && cd "$os" || exit

    # Fetch RSS feed, extract download links, and download the files
    curl "https://sourceforge.net/projects/pinn/rss?path=/os/$os" \
    | grep "<link>.*</link>" \
    | sed 's|<link>||;s|</link>||' \
    | while read url; do
        # Clean up the URL (removes /download part)
        url=$(echo "$url" | sed 's|/download$||')
        # Download the file
        wget "$url"
    done

    # Go back to the parent directory
    cd ..
done
```

- i hope bundling two pictures is a placeholder, clearly showing that update parsed, not a real space saving solution :D

procount, 2024-10-11:
I'm really sorry, but I'm not understanding your issues, or how any of these relate to PINN. Maybe something is getting lost in translation...?

> My external drive turns off each boot when screen splashes, even being connected to the external power source via the cable

So I think here you are commenting that your external drive powers down temporarily after you have booted PINN  and just before you boot into one of your installed OSes...? This is normal for any drive as the Raspberry Pi will reboot to boot into one of its installed OSes. No driver will help avoid this.

The rest seems maybe to do with Android? If you have problems with the OSes you need to contact the OS maintainers. I only convert the format of the OS so that it can multiboot.

andriilive, 2024-10-11:
Sorry for the earlier message, I was drafting on the go. Hopefully,  its is clearer after update. Just sharing a few bugs from pi-imager version 3.9.2, gaps in the manuals and a mess the noobs can face 

> You mentioned that your external drive powers down temporarily after booting into PINN.

No, it was powering down when the colored video signal was sent, before entering recovery mode. The issue disappeared later without any changes in PINN settings. It seems the TV was turning the Pi off and on using CEC. I had previously disabled CEC on a TV, but I hadn’t hard-rebooted it to apply the changes :/ so continued investigating false trail which brought me again to USB ports check 😑