procount/pinn #811: no freespace check before update

pilou75, 2024-05-11:
There is no freespace check before update.

I had installed PINN which was updated several times, and finally 3.8.8. Partition size was 64MB.
I was asked to update to 3.9, it ran out of space and was unable to boot.

procount, 2024-05-11:
It is very tight on space, only about 1MB free at the end. Did you have some extra files stored on your boot partition? 
It's difficult to check how much free space is needed because many of the files are overwritten. I looked at it more from the perspective of PINN's size being smaller than the partition size, and if this is not the case then the previous installation will be restored. I guess this check of freespace could be improved, but all my upgrade tests of a standard 3.8.8 to 3.9 version worked ok.

Meanwhile to restore PINN in a working fashion, you can download https://sourceforge.net/projects/pinn/files/pinn64/pinn-lite.zip/download and unzip it to your PINN partition. You will have to exclude wallpaper.jpg and rename wallpaper2.jpg to wallpaper.jpg.
At some stage you WILL need to create a bigger RECOVERY partition as described int he release note.

hahnmt, 2024-05-12:
Just booted and pulled down upgrade.  After upgrade partition filled leading to Kernel Panic on boot.  Partition size is 64MB.

pilou75, 2024-05-12:
Same as hahnmt commented.
I removed those directories.
I will see what I can do to expand recovery without reinstalling everything

procount, 2024-05-12:
I have temporarily blocked self update until I can do better disk space checks.
It might be easier to restore a working disk by deleting the contents of the recovery partition and then unzipping pinn-388.zip onto it. That way rhere will be no diskspace issues.

procount, 2024-05-15:
Self-update to 3.9.1 has been re-enabled.
It now wipes the whole of the boot partition before installing the new version to ensure full capacity is available. Settings will still be preserved.

procount, 2024-05-16:
Wpa_supplicant.conf is imported from the boot partition, so once imported it is no longer required there. Same for dhcpcd.conf