procount/pinn #812: video not properly initialized after recalbox 9.1

pilou75, 2024-05-11:
I don't know where the issue is. Recalbox or PINN.

When recalbox is used, and restarted, there is no display at PINN boot selection menu.

At first, I thought it was hanged. It's not, it's just a blank screen, otherwise looks OK.
I tried also with new install, and it's same. If recalbox 8.1.1 is used, no issue, but as soon as it has been upgraded to 9.1, blank screen.
Others distro I use, have no issue (raspbian, libreelec, batocera)

procount, 2024-05-11:
No idea either.
Sounds like something is not being totally reset on a soft reset. Probably something for RPL engineers to advise on :wink:

Are you using a Pi5? If so, try commenting out the dtoverlay=vc4-kms-v3d line in config.txt of PINN to see if it makes a difference. 🤷‍♂️

pilou75, 2024-05-12:
I forgot to mention. It's a PI4B

pilou75, 2024-05-18:
well, I updated Libreelec from 10 to 12, and now, I don't have any display when (warm) reboot 
somebody else have similar issue ?

JanSykor, 2024-05-21:
I have similar issue when I updated firmware on raspberry with pinn in Libreelec 10, sudenly I have "black" screen when booting LibreElec 9.2.8
For me help copy of file `start.elf` and `fixup.dat` from booting partition of LE10 to booting partition LE9.2.8
https://forum.libreelec.tv/thread/25514-libreelec-rpi4-arm-9-2-8-img/
Maybe it is also something with firmware.
Jan

pilou75, 2024-05-22:
Short answer, does the same problem occur with RaspberryOS no, does the same problem occur with LibreElec yes

Here is all tests I did today : 

LibreELEC 12 without PINN : the rainbow screen appears shortly during reboot = looks good to me
LibreELEC 12 with PINN : "blind" screen after reboot, I mean no rainbow, no press-shift picture, no boot selection menu, nothing until KODI starts

RaspiOS is always fine (rainbow screen appears shortly), with or without PINN

I have several things attached to the PI4 : USB Keyboard, USB PS3 Controller, and Hifiberry digi.
I tried to remove them, one by one, no improvement.

I would to try using another display (LG TV), but I have nothing else at the moment.

pilou75, 2024-05-23:
> I have similar issue when I updated firmware on raspberry with pinn in Libreelec 10, sudenly I have "black" screen when booting LibreElec 9.2.8 For me help copy of file `start.elf` and `fixup.dat` [...]

I tried with recalbox : 8.1.1 (PINN) is OK, but after upgrade to 9.1 have the issue.
I put back bcm2711-rpi-4-b.dtb, start.elf, and fixup.dat from 8.1.1. It doesn't help.

pilou75, 2025-02-21:
I have upgraded to 3.9.3 - although there is now a font size issue (which looks to be fixed in 3.9.4 - but not yet available as auto update) - the issue is not here anymore. 
I can confirm it's fine for example with recalbox (which have not been updated at all). 
I close the issue