procount/pinn #813: Raspian Jessie won't install from PINN

rickmoon1, 2024-05-15:
Hello procount, firstly thanks very much for a great product. I've used it successfully on my RPi 4 and I like the ability to be able to backup an OS. 

I'm having trouble getting an old version of Raspian to run on my old Rpi 3 from a just updated (v3.8.9) version of Pinn Lite. It looks like it's about to start, but then the script reveals a Kernel panic error. Running the fix utility reveals script errors on the partition_setup.sh file and says it needs to be updated, but I'm not sure where to get a suitable updated version. The version of Raspian Jessie was lifted from NOOBS v1.9.3. I also have installed LibreELEC v12 which I know will run from PINN, but haven't attempted in this instance to do that until I have sorted the Raspian Jessie issue.

Details are
Hardware - RPi 3 Model B rev 1.2
Software - PINN Lite 3.9.1, Raspian Jessie (2016-09-23), LibreELEC (12.0.0)
OSs installed from USB drive as opposed to the SD card

The Raspian and LibreELEC install and run fine on the same RPi on NOOBS 1.9.3.

I attach the PINN log file (added .txt extension) and screenshot of the Raspian boot freeze

Many thanks
rickmoon1
[image]
[log.txt](https://github.com/procount/pinn/files/15321751/log.txt)

procount, 2024-05-16:
OK I'll see what I can do, but why do you need Jessie? - it's quite old now.

rickmoon1, 2024-05-17:
I tried Bookworm but it was very slow, so I thought I'd go back to the version on the Pi 3 starter kit NOOBS card. I haven't tried Stretch, but I see it's on your supported OS list. 

I lifted Jessie straight from NOOBS 1.9.3 os folder, so would doing the same from a suitable verison of NOOBS work for Stretch ? I'm happy to do that and see if it goes OK on the Pi 3 to save you the unnecessary effort to get Jessie to work.

lurch, 2024-05-17:
You might want to try Bullseye, as this was the version immediately prior to Bookworm, and is still listed as the Legacy OS on https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy

procount, 2024-05-17:
PINN still includes a version of Buster in its Legacy category from 2023. That would be the easiest to test immediately. I still need to update it to Bullseye.

rickmoon1, 2024-05-18:
Hi all, tried a version of Stretch lifted from NOOBS and it's fine. Might be interesting to see how the Pi 3 copes with Buster or Bullseye. Thanks all for your input. Best wishes, rickmoon1

procount, 2024-05-19:
Raspios Bullseye has been included in the Legacy category

closed 2024-05-22 by procount