procount/pinn #814: include libreelec for kodi 20 on pinn

BENDER35, 2024-05-16:
hello

I comment on the topic, I have already released the update for Kodi 21, but in a group of an addon that I use (which I am not going to say), they tell me that the version for Kodi 21 is going to take a while, so my question is, will it take? Could you include the latest version of libreelec compatible with kodi 20 in pinn? Is it so that if the installation of 20 that I have (I have a raspberry pi 3) fails, it could be easily restored.

All the best

Rodrigo

procount, 2024-05-16:
So in summary, if I understand you correctly, you want me to include the previous version of LibreElec that used Kodi 20, in addition to the latest version, so that you can use a particular addon?

BENDER35, 2024-05-16:
> So in summary, if I understand you correctly, you want me to include the previous version of LibreElec that used Kodi 20, in addition to the latest version, so that you can use a particular addon?

hello

The addon that the creators say will take a while to become compatible with Kodi 21 is Moe's TV (included in Luar), an addon that allows you to watch Spanish on-demand television and live channels.

Greetings

procount, 2024-05-16:
OK I'll see what I can do.

procount, 2024-05-19:
LibreElec v11 has been restored in the Legacy category

closed 2024-05-19 by procount