procount/pinn #817: Update from 3.9.1 to 3.9.2 failed

JanSykor, 2024-05-26:
Hello @procount 
I have update failed message on RPi3 and RPi4.
Here is picture of log:
[image]
If you need aby other detaily, please let me know.

On Pi5 update was OK.
Thanks
Jan

procount, 2024-05-26:
Thanks for the report.
It's failed because there is not enough space on the RECOVERY partition for the new version. About 1.5MB less. Not sure why - did you have some additional files on there?  Could you list what is left on there now (ls -alF /mnt)?
I tested the rollback if there is not enough space, but your rollback failed for some reason. 
(If you haven't rebooted, you can copy your files from /settings/mnt to /mnt.)

JanSykor, 2024-05-27:
Hi @procount 
Here is logs and file list from RPi3
Rollback not failed - installation was not damaged and I have still 3.9.1 version - just installation failed.
[logs.zip](https://github.com/procount/pinn/files/15451183/logs.zip)
No effect depend keyboard - same result in english nor czech
Thanks
Jan

mbernardi1961, 2024-05-30:
I have the same issue using a RPi4.

procount, 2024-05-31:
Updates from 3.9.1 to 3.9.2 should now work ok, if you'd like to retry it.

JanSykor, 2024-05-31:
Nice job @procount . Thank you
Small wallpaper selected and language was necessary select again (english and then again czech) to apply last language pack.
Working perfect now on Pi3 And Pi4.
Jan

closed 2024-05-31 by procount