procount/pinn #818: download speed very slow Indian mirror sourceforge.net

naidu72, 2024-05-29:
is there any way to set mirror url  while doing the setup or configuration in PINN os installation

procount, 2024-05-29:
Not yet.
For now you could download all the required files manually from sourceforge to a USB drive. From their webpage you can select a mirror. Then install the OS locally from your USB drive. 
See README_PINN.md for details.

naidu72, 2024-05-29:
if i want install multiple operating systems  will it be possible from usb? if yes please share me reference url  for local USB drive

procount, 2024-05-29:
Yes, you can install multiple OSes from USB drive.
Please see https://github.com/procount/pinn/blob/master/README_PINN.md#local-images-on-usb-stick and also look at https://github.com/procount/pinn/blob/master/README_PINN.md#download for some background information, although I don't recommend using the download option in your case since there is still no mirror selection there. Instead go direct to the source of the OS on sourceforge or wherever. e.g start at https://sourceforge.net/projects/pinn/files/os/

Choose an OS and copy the folder and all of its subfolders and contents. If a file is slow downloading, click the "problems downloading" button and select a different mirror. Put the folder and its contents on a FAT32 formatted USB drive in a /os/ folder. If you want to install kde3lee64, you should end up with /os/kde3lee64/ with all of its contents and subfolders stored within it on your USB. Put other OSes in their own folder under /os/. Then boot PINN with the USB drive inserted and PINN will find the OSes.

The above is just an example.
PINN uses many repositories. To view them all, start at https://github.com/procount/pinn-os/blob/master/os/repo_list.json and follow the chain to the OS you need.

naidu72, 2024-06-01:
i really appreciate for your support, yes i am able install from  local USB as offline mode installation.
--------------------
i have another two question in my head 
Question1:
 in PINN boot menu has replace option, can we replace installed OS with new OS from the local USB?

Question 2:
in case of OS upgrade,  example currently LineageOS 21 has  (Feb 2024) build if they  releases new version say this month. how that upgrade can be done ?

closed 2024-06-18 by naidu72

procount, 2024-06-18:
I have no control over the download speed :wink:

Sorry, I missed your previous questions:
1. Yes. It does not matter where the new OS comes from. It needs to have the same partition layout and fit within the allocated space.
2. Refer to the OS upgrade instructions. In most cases it won't make a difference that it is installed under PINN, but there may be some exceptions.