procount/pinn #820: GAPPS installation is failing on lineage21-rpi4

naidu72, 2024-06-18:
I able to install and boot lineage21-rpi4 on PINN multi boot. when i am trying to install "MindTheGapps-14.0.0-arm64-xxxxxxxx_xxxxxx.zip" during the TWRP recovery boot it is failing for installing the zip

procount, 2024-06-18:
I only take care of converting the format of Lineage to make it multi-bootable.  Installing MindTheGapps has worked in previous versions where I needed to use it, but I've not tested it in this version.
I suggest you try installing Lineage without PINN and check MindTheGapps installs normally, If that fails you can reach out to Konsta or the author of MindTheGapps. Konsta won't help with PINN installations and I know little about MindTheGapps or Lineage.

naidu72, 2024-06-18:
Thanks for your response.
i will try your recommendations

closed 2024-06-18 by naidu72