procount/pinn #822: Ubuntu 24.04

ZAIAC001, 2024-06-21:
**Describe the bug**
Kernel panic at boot
(invalid magic at start of compressed...)
23.10 works fine

**To reproduce**
Fresh install or update from 23.10

**System**
Pi 5 8gb + NVME
PINN 3.9.2

procount, 2024-06-21:
Are you able to try the following tests to narrow the problem down? 
1. Install Ubuntu 24.04 to an SD card with RPi Imager.
2. Install PINN to an SD card and use that to install Ubuntu 24.04 to the same SD card
3. Install Ubuntu 24.04 direct to the NVMe (You may boot some OS off your SD card and install RPi Imager to it to install Ubuntu to the NVMe).

ZAIAC001, 2024-06-22:
Install Ubuntu 24.04 to an SD card with RPi Imager partial works
It runs install screens but soooo slow
And after complete this screens drop unsolvable error with user creation
User not created
Login blank 
Need reinstall again (not sure that it help me)

procount, 2024-06-22:
I forgot that there was an issue with the official Ubuntu 24.04 version. Please see https://forums.raspberrypi.com/viewtopic.php?p=2227735#p2227735 for a link to the img file I used for PINN.

I was trying to determine if your problem was to do with the img, your NVME drive,  PINN or your process.

ZAIAC001, 2024-06-23:
Well as I guess ubuntu img corrupted in source
Im sorry for disturb
Сan you write to me when good img will be available in PINN

closed 2024-06-23 by ZAIAC001

procount, 2024-06-24:
Now that I have an NVME drive, I can confirm the 24.04 image in PINN does not install well to an NVME drive, although it worked ok on an SD card. The official Ubuntu 24.04 does not fare much better on my NVME drive.

ZAIAC001, 2024-06-25:
Is there any guide how to create PINN install folder with OS from usual img?
There are some OSes that I want to install
But they're not listed in PINN

(For fools preferable)

PS: I can follow instructions and copy commands in terminal ))) 
PSS: I have SD card and USB stick and NVME (where currently ubuntu and raspios installed)

procount, 2024-06-25:
There is no guide that will work for all OSes as each OS has its own requirements and constraints. Each new OS I convert can be a new experience, although most are quite similar.
However, you can read some general instructions in the main documentation at https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-create-a-custom-os-version. That contains a link to an example tutorial in my Wiki too.

androot2, 2024-07-21:
Hi, I have the same issue. Similiar configuration, but less RAM on Raspberry. With previous version of Ubuntu was the same peoblem

procount, 2024-07-21:
Ubuntu 24.04 has had several issues with RPi5.
Iirc, the pinn version now works ok on SD card, but doesnt work well on NVME SSDs. Ubuntu maintainers need to fix this. I can't do much about it.

I have modified Ubuntu 23.10 to include the initrd from an earlier version of Ubuntu 24.04 to get it to boot on NVME SSD. This works fine for me. If you are having problems with it  please first confirm that a plain install of the same Ubuntu 23.10 version works when installed with Rpi Imager.
I would need to know exact details of your hardware (less ram is not very precise) and how you installed it to help identify your issues, but I'm mostly limited to problems related to PINN and OS conversions rather than issues with an OS itself.

ZAIAC001, 2024-07-21:
> Hi, I have the same issue. Similiar configuration, but less RAM on Raspberry. With previous version of Ubuntu was the same problem (few months ago). How to fix it?

Yeah. We have already discovered this issue. The problem is with Ubuntu itself. There's nothing to do yet
So leave @procount alone for now

Currently I'm using fydeos
(It's a pity that pinn doesn't support this OS. But there is nothing to do)

androot2, 2024-07-22:
Thank you!

23.10 works fine, after you modified it. I didn't know that, 24.04 won't work on my setup. Pi5, 4GB, custom made NVME extension. Other OSes works well. Kodi on LibreElec, Raspbian.

ZAIAC001, 2024-07-22:
> Thank you!
> 
> 23.10 works fine, after you modified it. I didn't know that, 24.04 won't work on my setup. Pi5, 4GB, custom made NVME extension. Other OSes works well. Kodi on LibreElec, Raspbian.

Well, Ubuntu 24.04 is only one OS that wont work on nvme. (That I try to install)
Even 23.10 without any manipulations works fine (until it updates to 24.04 by its own)

ZAIAC001, 2024-07-22:
> custom made NVME

Can you describe your custom nvme hat?
Im looking for nvme hat with two nvmes
And I want to switch between them by some switch or button (on power off state)

Now Ive got a china made hat
But thinking about make my own pcb

androot2, 2024-07-22:
Hello

I based on someone's project and made my own. In attached file you can find both.
[Pi5_M2_Hat.zip](https://github.com/user-attachments/files/16331904/Pi5_M2_Hat.zip)

If you decide to use it, it's one mistake on this board. RESET signal shouldn't be inverted.

procount, 2024-11-20:
Ubuntu boot panics on NVME should now be resolved.