procount/pinn #823: Touchscreen 15 inch

ZAIAC001, 2024-06-23:
**Describe the bug**
Touchscreen offset appears in PINN
(In systems everything is normal)
Offset seems random on each tap

**To reproduce**
Just connect touchscreen

**System**
Pi 5 8gb + NVME + Touchscreen (Via hdmi and type-c for data)
[Link to my screen on aliexpress](https://aliexpress.ru/item/1005004495109392.html?spm=a2g2w.orderdetail.0.0.64a04aa6HvgSnH&sku_id=12000033880937533&gatewayAdapt=glo2rus)

ZAIAC001, 2024-06-23:
Im trying to make some kind of tablet and it would be great if this problem will be solved

ZAIAC001, 2024-06-23:
Maybe I need some drivers in PINN?

procount, 2024-08-20:
You would need to identify which drivers are required for your touchscreen e.g. in Raspios.
Use something like `lsusb` before and after plugging in the touchscreen to identify which drivers are used.

Numidara, 2025-01-08:
I have a similar problem for a touchscreen on RASPAD3 : 
https://www.sunfounder.com/products/raspberrypi-tablet-raspad?variant=37227248386209
Any fix ?

procount, 2025-01-08:
PINN is very small and only supports a limited set of touchscreens (Raspberry pi 7" & Hyperpixels), and even then, only for models before the PI5 (i.e. no KMS support yet).
Probably the RASPAD3 needs different touchscreen drivers.
I don't have access to a RASPAD3, so you would need to identify which touchscreen device you are using, what drivers it is loading up, and any specific configuration needed in config.txt to enable or configure it (if necessary).

If the touchscreen works ok in Raspios, have a look in the output of `dmesg` and `lsusb` (if connected via USB) for any mention of the touchscreen to identify it. 

I can then look to adding the correct drivers.

Numidara, 2025-01-08:
222a:0001 ILI Technology Corp. Multi-Screen ;)
https://linux-hardware.org/?id=usb:222a-0001

Bus 003 Device 004: ID 222a:0001 ILI Technology Corp. Multi-Touch Screen
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x222a ILI Technology Corp.
  idProduct          0x0001 Multi-Touch Screen
  bcdDevice            0.02
  iManufacturer           1 ILITEK
  iProduct                2 ILITEK-TP
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              400mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     850
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

Thank you in advance :)