procount/pinn #824: Cannot force TV display resolution for PINN 3.9.2

Hiforever, 2024-06-25:
I recently upgraded to PINN 3.9.2 on my RP4 8G which never detect my LG 4k TV correctly ever since.   In the past, I force it to display the desired resolution using the following lines in Config.txt.   I understand that this no longer works for PINN 3.9.2

[HDMI:0]
hdmi_group=1
hdmi_mode=95
 
[HDMI:1]
hdmi_group=1
hdmi_mode=16

I then tried appending the following into the first line of my cmdline.txt (after "sdhci.debug_quirks2=4") but it didn't seems to work.  Did I miss anything ?  

video=HDMI-A-1:1920x1080M@60D

procount, 2024-06-25:
Only the Pi5 uses KMS under PINN and uses the cmdline.txt video configuration (if required).
PINN 3.9.2 by default still uses the old 5.10 kernel for the Pi4 and below and KMS is not enabled, so the old config.txt settings should still work.
Of course, you can alter config.txt to force the Pi4 to use KMS and the 64-bit kernel if you want to.

If you boot into your Pi4 with no storage devices so that you get the diagnostic screen, does it indicate a valid EDID can be retrieved?

procount, 2024-06-25:
You could also try:
```
hdmi_group:0=1
hdmi_mode:0=95

hdmi_group:1=1
hdmi_mode:1=16
```

Hiforever, 2024-06-26:
Include the hdmi_group lines in config.txt as advised but it didn't work.

BTW, I believed that KMS is enabled though I never modified the config.txt.  My PINN 3.9.2 is flashed from PI Imager.  Below is the config.txt hat I got once PINN is installed.  

gpu_mem=16
start_file=start.elf
fixup_file=fixup.dat

disable_overscan=1
initramfs pinn.rfs

[HDMI1]
hdmi_force_hotplug=1

[pi4]
start_file=start4.elf
fixup_file=fixup4.dat
max_framebuffers=2

[board-type=0x17]
kernel=kernel8.img
overlay_prefix=overlays6/
dtoverlay=vc4-kms-v3d
max_framebuffers=2

procount, 2024-07-04:
With the above config.txt file, then the Pi4 will NOT use KMS, it uses the older firmware drivers.
The `[board-type=0x17]` section is for the Pi5. If you delete it and it stops working, that suggests you have a pi5.... :~
Are you sure you have a Pi4 not a Pi5?
Also I suggest using HDMI0 (the one next to the USB-C power connector) instead of HDMI1

Hiforever, 2024-07-09:
Thanks for your advice.   I confirmed that I am using Pi4 8GB instead of Pi5.   I tried different config.txt setting in the last few days with results as follows 

1.   The crash is casued by [HDMI:0] and [HDMI:1] regardless of their position with error message is
      ioctl FBIOPUT_CON2FBMAP: Invalid argument
      
     This is resolved using 
     hdmi_group:0=1
     hdmi_mode:0=95

     hdmi_group:1=1
     hdmi_mode:1=16

     I'm not sure if I can use [HDMI1] & [HDMI0] instead.

2.   It seems that any lines below the [board-type=0x17] section will not work.   Originally, I placed your recommended hdmi lines (above) after the [board-type=0x17] section but it has no effect.    This is the reason why I beleived KMS is being used when I rasied this issue.   Now I put everthing before the [board-type=0x17] section and it works.

procount, 2024-07-09:
The lines with square brackets are the start of separate conditional sections that only apply if the condition is true.
See https://www.raspberrypi.com/documentation/computers/config_txt.html#conditional-filters.
Everything after [board-type=0x17] only applies to the Pi5.
You need the [pi4] section.
See also https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#legacy-conditional-filters

Glad you got it working.

closed 2024-07-25 by Hiforever