procount/pinn #825: PINN won't boot without dispaly

ZefiroHST, 2024-06-30:
**Describe the bug**
I succesfully set up a pi5 with PINN and a couple of OSes using a keyboard and a display. I want to use it now in an headless configuration. If I turn on the pi5 without the display attached looks like PINN is not able to boot.

**To reproduce**
I have a pi5, 8GB. I am booting from an ssd nvme Kingstone 1TB NV2 connected via the Argon NEO 5 m.2 nvme PCIE Expansion Board. I have a fresh installation of PINN 3.9.2 (built may 22). I flashed PINN on the ssd with the official raspberry pi imager. I succesfully installed some OSes via ethernet connection. I haven't modified anything in config and cmdline files. The display I am using is an HP monitor connected via HDMI cable. 

**Expected behaviour**
I have also a pi4, 8GB with PINN (I can not confirm the version, but it is an older one. I set it up in 2022) booting from an SSD connected via a sata to usb 3.0 adaptator. When I turn on the pi4 the last selected OS boots independetly of the presence of a display. I would expect the same with the pi5. 

**Actual behaviour**
I am able to verify that when I turn on the headless pi5 the OS which should boot after 10sec does not boot (I can not ssh and the crontab I setted up is not starting). If I connect a display after a while I can see only a black screen. If I start the pi with the display connected and turned on I do not experience any issue. If I disconnect the display right after PINN boots (while I see the press shift message) the pi proceed booting the last OS selected without any issue (I can ssh and crontab task are done).

procount, 2024-06-30:
This is probably due to Pi5 requiring KMS for the video and it not finding an edid when no monitor is attached.
I'm not able to look at this for the moment, but I suggest firstly in config.txt to duplicate the hotplug setting for hdmi0. Then in cmdline.txt add an appropriate video setting such as "video=HDMI-A-1:1280x800@60" to see if either of those work for you.

ZefiroHST, 2024-06-30:
I tried what you suggested. Nothing changes. Here is the precise files I used. 

**config** 

gpu_mem=16
start_file=start.elf 
fixup_file=fixup.dat

disable_overscan=1
initramfs pinn.rfs

[HDMI0]
hdmi_force_hotplug=1

[HDMI1]
hdmi_force_hotplug=1

[рі4]
start_file=start4.elf 
fixup_fle-fixup4.dat
max_framebuffers=2

[board-type=0x17] 
kernel=kernel8.img
overlay_prefix=overlays6/ 
dtoverlay=vc4-kms-v3d
max_framebuffers=2

**cmdline**

repolist=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json quiet ramdisk_size-65536
root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline loglevel=2 sdhci.debug_quirks2=4 
video=HDMI-A-1:1920x1080M@60 video=HDMI-A-2:1920x1080M@60 ssh

procount, 2024-07-01:
Try adding `vncinstall` (NOT vncshare) to cmdline.txt.

procount, 2024-07-03:
This is due to the KMS video driver that is used on the Pi5.
Previously, the firmware always created a framebuffer /dev/fb0. The Qt framework requires a framebuffer to work.
However, in KMS, framebuffers are created dynamically when it detects a monitor is physically attached. No monitor, no framebuffer, so Qt crashes.

Adding `vncinstall` works because it uses VNC as the only framebuffer. `vncshare` doesn't work because Qt also expects an additional framebuffer. As it doesn't exist, it crashes. `vncinstall` is not so good because if you do attach a monitor you won't see much as the display is only visible then over VNC.

A better solution is to specify the video resolution in cmdline.txt
e.g. Add `video=HDMI-A-1:1920x1080M@60**D**
Note that the 'D' is important as this force the framebuffer to be created even when no monitor is present.
This setting will override your monitor's EDID data, so choose a resolution that matches your monitor so that when it is attached, the display will look the same.

Also note that using PINN under Pi5/KMS is still a work in progress so some things still don't work, like DSI/DPI displays.

closed 2024-07-03 by procount

ZefiroHST, 2024-07-04:
I tested the second method suggested, it works perfectly. Thanks for the amazing support and work done so far!

k2helix, 2024-09-21:
I tried the second suggested method and my RPi 5 behaviour is the following:

- If there is an HDMI cable connected to another HDMI device that is turned on, it boots correctly.
- If there is an HDMi cable connected to another HDMI device that is turned off (without power), it boots correctly.
- If there is an HDMI cable connected to nothing, it won't boot
- If there is no HDMI cable, it won't boot.

It seems really weird. Any idea why this could be happening? I sometimes use it with a display and sometimes without it, so any help would be appreciated.

procount, 2024-09-21:
Yes, only the first line of cmdline.txt is processed.

UserX404, 2024-11-16:
@k2helix
Did you add the HDMI setting only or also the vnc stuff?

k2helix, 2024-11-16:
> @k2helix
> Did you add the HDMI setting only or also the vnc stuff?

I added what was recommended as a better solution: `video=HDMI-A-1:1920x1080M@60D`

UserX404, 2024-11-16:
I'm using RPi5 with PINN and Libreelec where i found someting about [EDID](https://wiki.libreelec.tv/configuration/edid) which adds 
```
drm.edid_firmware=HDMI-A-1:edid/edid-HDMI-A-1.bin video=HDMI-A-1:D
```
to cmdline.txt automatically.
Still not working atm but maybe a path issue.

lurch, 2024-11-16:
Funnily enough, some of my Raspberry Pi colleagues were talking about custom EDID files just the other day... apparently a custom EDID file won't work if you're using an initramfs (which I believe PINN is), because by the time the kernel is able to load the EDID it's already "too late" as the DRM driver has already been initialised.

saaraswata, 2024-12-26:
Can confirm the second method works 😀