procount/pinn #832: headless bookworm install, no wifi setup

narspt, 2024-08-20:
Hi, I used to use PINN to easily manage and reinstall OS's on completely headless RPi's, it always worked great and Wifi used to be setup automatically by PINN, however since Bookworm this seems not done anymore, I did check and it seems Bookworm doesn't accept wpa_supplicant.conf anymore, I guess this is why PINN can't setup Wifi details anymore, however there must be another way to do it (I don't know yet how) as the official RPi USB Installer apparently allows to setup Wifi details, would it be possible you get this working for PINN also to automatically setup Wifi as it used to do with older OS's?

procount, 2024-08-20:
Since the introduction of bookworm and the Pi5, many things have unfortunately broken.
Wifi setup only worked with wpa_supplicant, but with the change to network manager this has broken too. Once I have figured out how to convert, I will adapt it. For now it needs some manual intervention.

narspt, 2024-08-20:
Thanks for your reply, I understand. Btw, maybe the following may be interesting for you (if you didn't see it yet):
https://github.com/raspberrypi/bookworm-feedback/issues/72
there are some workaround methods mentioned there, like the one with firstrun script, apparently the method used by rpi-imager.

Btw, for now I did it manually, connected ethernet cable and used these commands via ssh:
```
sudo nmcli radio wifi on
sudo nmcli dev wifi connect <wifi-ssid> password "<network-password>"
```

Ircama, 2024-12-11:
This is the way I performed the initial WiFi and ssh configuration of a headless system after installing the Bookworm version of Raspberry Pi OS via PINN:

- After the initial installation of the selected OS with PINN, reboot to PINN (VNC and rescue shell via ssh)
- Login to the *rescue* shell via ssh (`ssh root@192.168...` with password `raspberry`)
- Run the following commands:

```bash
umount /mnt
mount /dev/mmcblk0p7 /mnt
chroot /mnt
mount -t proc proc /proc
/usr/lib/raspberrypi-sys-mods/imager_custom set_hostname <my hostname>
/usr/lib/raspberrypi-sys-mods/imager_custom set_wlan <SSID> <password>
/usr/lib/raspberrypi-sys-mods/imager_custom enable_ssh
raspi-config nonint do_change_pass
exit
umount /mnt/proc
umount /mnt
```

Continuing with the *rescue* shell, enable the OS to start at boot (instead of VNC/ssh to PINN):

```bash
sudo mount /dev/mmcblk0p1 /mnt
echo "boot_partition=6" > /mnt/autoboot.txt
sudo umount /mnt
sync
```

- Reboot
- Connect via ssh (`ssh pi@192.168...`)
- Via `sudo raspi-config`, activate VNC
- Connect to the system with VNC and continue the installation procedure

narspt, 2024-12-12:
Thanks for sharing your steps @Ircama !