procount/pinn #834: Raspios not showing latest version (2024-07-04)

Hordeking, 2024-09-01:
**Describe the bug**
PINN Lite offers the Raspios version from 2024-03-15, but does not offer 2024-07-04
https://downloads.raspberrypi.com/raspios_armhf/images/raspios_armhf-2024-07-04/2024-07-04-raspios-bookworm-armhf.img.xz
https://downloads.raspberrypi.com/raspios_full_armhf/images/raspios_full_armhf-2024-07-04/2024-07-04-raspios-bookworm-armhf-full.img.xz
https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2024-07-04/2024-07-04-raspios-bookworm-armhf-lite.img.xz
...And so on.

**To reproduce**
No special steps.

**Expected behaviour**
Installer should offer updated version.

**Actual behaviour**
Out of date version offered

**System**
* Raspberry Pi 3B
* PINN 3.9.2

**Additional context**
Worked on trying to produce my own OS metafiles for inclusion on USB stick based on the ones produced by downloading the offered version. Was successful at separating Raspios image into boot.tar.xz and root.tar.xz, and determining most items in os.json. Was unable to determine how download_size is computed and unable to find clear instructions for how to do this manually.

What do I need to do to produce the correct value for this field?

Hordeking, 2024-09-01:
Posted Update request at [https://github.com/procount/pinn-os/issues/20](https://github.com/procount/pinn-os/issues/20)

closed 2024-09-01 by Hordeking

procount, 2024-09-01:
> Was unable to determine how download_size is computed and unable to find clear instructions for how to do this manually.

Try the following links:
- https://github.com/procount/pinn/wiki/JSON-fields#download_size-integer-o
- https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN
- https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-create-a-custom-os-version

Hordeking, 2024-09-01:
That's helpful. Especially the explanation of the json fields. No clue why that one didn't come up in the searches. Thanks.