procount/pinn #835: how can I make NOOBS support RPI5

devbzwlyukfkz5t, 2024-09-06:
Thanks for your project, how can I make NOOBS support RPI5, (PINN does not support built-in OS files to SD card)

Thank you.

procount, 2024-09-06:
I'm not sure I understand your question as PINN supports everything that NOOBS did as it is a fork of that project. Can you clarify exactly what you want to do?

devbzwlyukfkz5t, 2024-09-06:
> I'm not sure I understand your question as PINN supports everything that NOOBS did as it is a fork of that project. Can you clarify exactly what you want to do?

Thanks for the reply, I was hoping to put rpios64 and rpios32 or some other OS on the SD card with NOOBS or PINN pre-installed instead of downloading from the server.

like this:
-NOOBS_v3_8_1
--os
---LibreELEC_RPi
---LibreELEC_RPi2
---LibreELEC_RPi4
---RaspiOS_64bit
---RaspiOS_Full_armhf
---Other systems

procount, 2024-09-06:
This is still possible.
I just stopped providing a FULL version because
- it takes too long to produce
- it wastes a lot of sd card space
- there were not many downloads
- users want to include different OSes
- i would propose storing your OSes on a separate USB drive instead to give you more space on your sdcard for the installed OSes. It is also more flexible.
However, if you still want to create such a version, I can provide instructions.

devbzwlyukfkz5t, 2024-09-06:
> This is still possible. I just stopped providing a FULL version because
>
> * it takes too long to produce
> * it wastes a lot of sd card space
> * there were not many downloads
> * users want to include [...]

I will try to use PINN to pre-install the complete system. I would also like to know how NOOBS needs to be modified to support RPI5. Thanks for your guidance.

I compared the NOOBS and PINN files and found the main differences:
- bcm2712-rpi-5-b.dtb, rpi5 boot file?
- Is board-type=0x17 in config.txt for compatibility with rpi5?

procount, 2024-09-06:
> I will try to use PINN to pre-install the complete system.

Please wait a short while until I can respond properly to avoid you doing rework.

How big is your sd card? Do you also have a spare usb drive?

devbzwlyukfkz5t, 2024-09-06:
> > I will try to use PINN to pre-install the complete system.
> 
> Please wait a short while until I can respond properly to avoid you doing rework.
> 
> How big is your sd card? Do you also have a spare usb drive?

256GB microSD card, 128 GB USB flash drive.

procount, 2024-09-06:
To save me the effort of duplication, I'm going to refer you to the User Documentation. If you haven't already found or read [README_PINN.md](https://github.com/procount/pinn/blob/master/README_PINN.md), I recommend it as it details all the additional features that PINN provides over NOOBS.

For your specific use case, please read the following sections:
https://github.com/procount/pinn/blob/master/README_PINN.md#local-images-on-sd-card
https://github.com/procount/pinn/blob/master/README_PINN.md#local-images-on-usb-stick
https://github.com/procount/pinn/blob/master/README_PINN.md#download

So in brief, you need to do the following, but see the above links for the details:
- Format your USB drive and create a folder called `os` on it.
- Install PINN to your SD card (I suggest using RPI Imager and install it from the Misc Utility category)
- Boot PINN on your SD card and insert your USB stick.
- Select all of the OSes you are interested in.
- Go to the Archive Menu and select Download, which will download them all to the /os folder on your USB stick.

Normally, this is where I would suggest you stop, as you can install all your downloaded OSes from the USB stick using PINN. All you need to do is plug in the USB stick when you need to install anything and the OSes will be available for installation.
However, if you want to put them on your SD card, I will put details in my next post.

procount, 2024-09-06:
If you want to copy the downloaded OSes from your USB stick to the SD card, you need to expand PINN's recovery partition to make room for them., but first we need to know how much additional space we need.
- Calculate the total Used Space in the /os folder of your USB stick. 
 
In Windows, you can just do FIle Properties in File Explorer on the OS folder. Remember to convert the figure to MB. In Linux (you can use PINN's recovery shell, or Raspios if you have another SD card with that installed) you can use `du -s -BM /path.to/os` (Replace path.to with the correct pathname. On PINN this might be /tmp/media/sda1, for example)

Make sure your figure is in MB and add a few MBs extra to make sure you will have enough space.

- Boot PINN on your SD card. 
- Go to the Maintenance Menu. 
- Make sure that the PINN entry is highlighted at the top of the list (it should be by default and is probably the only one)
- Click on Edit Config
- Click on Edit Options
- On the installation tab, check the option for runinstaller.
- On the row below that, check the reserve '+' box and enter the size of the /os folder that you calculated earlier in the box in MBs.
- click OK and OK again to get back to the Maintenance menu.
- Prexs ESC twice to reboot PINN

When PINN reboots, it will wipe any OSes already installed and resize the RECOVERY partition.
You can now copy the contents of the /os folder on the USB stick to the /os folder on PINN's RECOVERY partition. DO this however you want, but be sure to copy the whole folder structure.

You no longer need the USB stick and the list of OSes on the SD card will be presented next time you boot PINN. You can tell they are coming from the SD card due to the icon at the right hand side of each entry,

If you are connected to the internet, PINN will also download the list of other OSes available from there. If there are duplicates, PINN will only show the most recent version, which is likely is the internet version gets updated. If the dates are the same, the local entry will be shown in preference to avoid additional downloads.

procount, 2024-09-06:
>  I would also like to know how NOOBS needs to be modified to support RPI5.

This is not really an option. NOOBS is old, deprecated and no longer supported.
RPi5 needs a new 64-bit kernel. 
It does not support the "alternative recovery load mechanism" used in NOOBS. 
The buildroot version needed for the kernel does not support the version of QT used to build NOOBS.
RPI5 uses the newer KMS video drivers.
Font sizes are handled differently.
etc....

So it is not simplly a matter of adding a few extra files. It took me around 6 months to add RPi5 support to PINN and it is now usable but still not fully functional.
There is nothing that NOOBS can do that PINN can't do better.

devbzwlyukfkz5t, 2024-09-07:
Thanks for your reply. I have pre-installed the OS files into PINN.

closed 2024-09-07 by devbzwlyukfkz5t