procount/pinn #836: How to install the downloaded image?

paolosezart, 2024-09-06:
I downloaded the OS image from https://konstakang.com/devices/rpi4/LineageOS21-ATV . I installed PINN on the USB-SSD and copied the file lineage-21.0-20240816-UNOFFICIAL-KonstaKANG-rpi4-atv.img to the os folder on the USB-SSD. Just in case, I copied the zip file of lineage-21.0-20240816-UNOFFICIAL-KonstaKANG-rpi4-atv.zip to the os folder on the ssd disk. After downloading PINN on my raspberry pi 4B I didn't understand how to install the already copied operating system image. There is no option to install from disk in PINN menu. How to solve this problem? I would not like to use online installation, because my ISP slows down very much downloading files from https://sourceforge.net/ and https://github.com/ (on my computer I download via proxy at 180 megabits per second. While directly, the download speed is limited to 100-500 kelabits per second.
How can I solve this problem? Please write in detail how to install already downloaded images with PINN. Also, it would not hurt to specify in what format PINN “understands” operating system image files.

Thank you for your excellent development. 
I hope my problem is easily solved.

procount, 2024-09-06:
PINN does not install OS image files (*.img). These must first be converted by separating the file content of the image (from the partition layout) into *.tar.xz files. This is necessary because PINN needs to alter the partition layout in order to install multiple OSes. Each OS also needs additional meta files that tell PINN how to download and install it, and slideshow files displayed during the installation. So each converted OS comprises about 10 files stored in a folder, typically on sourceforge.net. So the usual operation of PINN is to install these OSes directly from the internet.

If this is too slow for you, you can download these folders onto a separate USB stick and then use PINN to install them from your USB stick to your SD card. PINN includes a download feature to do just this. However, if you need to download them manually via some other means, then you can do so. You just need to follow the chain of JSON files to find the correct URL of the OS folder you want, starting at: http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json.

I have just been answering another issue where my response is equally applicable to this question, so please refer to https://github.com/procount/pinn/issues/835#issuecomment-2333638372 for further details.

jmayone, 2025-03-23:
Appreciate this thread as there is not much information on duration of process.  Chiming in to report my experience and welcome feedback- RPi5 w/ 1TB SSD.  Selected images below in PINN and download and/or processing is .3 to 1.5 MB/sec.  I'm 3 hours in, and 13 hours remain.  Progress continues but the time only goes up.   This normal fam?    

**Images selected:**
Recalbox5 8.0.02 for RPi4
Moode 9.1.5
Raspios_arm64_full (64 bit port of Debian with desktop and recommended applications)

@procount - Really appreciate the knowledge you are dropping on this front.  I'm new to Pi, but will be seeing if I can action on some of your recommendations.  I don't mind waiting half day for PINN if it's a success 🤞

procount, 2025-03-24:
That really is quite slow. I'd be tempted to abandon it and do a reinstall in the hope you would get a faster mirror from sourceforge. I'm afraid I don't have any control over which sourceforge mirror you connect to. Some are faster than others. If you still get problems I suggest downloading them manually to a USB stick and then installing them from there. That way you can choose a specific mirror. 

The OSes are stored across several respositories, but most are stored in https://sourceforge.net/projects/pinn/files/os/ 
If you can't find the one you want there, then start by looking at https://github.com/procount/pinn-os/blob/master/os/repo_list.json and follow the links to the other repository lists.

You would need to download the whole of a folder of your OS of interest and all the files within it, to a USB stick. Put all of these folders under a folder called /os for them to be recognised by PINN. When you click on a file to download it, you should see a button on the webpage called "problems downloading". Click this and you can select another mirror to download from.

For more infomration, please see https://github.com/procount/pinn/blob/master/README_PINN.md#local-images-on-usb-stick

jmayone, 2025-03-27:
Greatly appreciate your reply which I've been struggling to get back to (along with project!)  The install ultimately wrapped up in about 9h.  I had trouble with alot of steps, no fault of anyone but my own incompetence on this front lol.  Since I'm such a noob, no pun intended, I have 3 simple questions for ya:

1- To set expectations, would I be able to use PINN (Select OS via touch) on waveshare DSI screen?
2- Do you have a link to donate to project?  I searched but could not find any, and the work you've done and support you provide across multiple forums is commendable. 
3- If 1 is a yes, are you for hire for direct assistance?*

*For some inspiration, I'm an oldschool hardware geek.  I love retrofits , have had a few featured, and my next project is retrofitting a 5" 'antique' portable TV made by Bentley.  I've got everything in house for the build at this stage which includes the Pi, screen, Dayton audio amplifier and some serious little speakers made by Tang Band.  It's gonna be super-sweet, but I'm feeling the pain on the multi-OS front.  Would love to use my 1tb NVM but I also may just use multiple drives.  As I plan to lead out all outputs including extending the microSD card.  In any case, really appreciate folks like you!

procount, 2025-03-28:
1. I have not added support for any waveshare screens. if it is compatible with the Raspberry Pi DSi Touch screen, or you can configure config.txt to work with this screen, then it might display something. The touch will probably require a new driver, but I would need to be informed about which one is needed.
2. No, but thank you.
3. It has been known, but not often. A possible change in architecture might be coming along. If it happens, it might make such support easier.