procount/pinn #837: Update to 3.9.1 seems to resize partitions and now the PINN partition seems excessively large.

rickwookie, 2024-09-08:
I know the PINN partition was previously too small to easily work with the new Pi 5 files, but doing an update seems to have left my PINN partition at 512 MB (double what it was before?), with 133 MB free space. How much space does it really need? Only now I can't update my LibreELEC installation at its partition (on my 8 GB SD card) seems to have been reduces and the "/storage" in LibreELEC is insufficient to unpack the update (for the Pi 5 image).

Is there a way to tweak the partition sizes without losing my data?

procount, 2024-09-08:
I can only guess you have some additional files on your RECOVERY partition, or maybe you have the `reserve` option set, or something strange is going on.

The original PINN RECOVERY partition was 64MB and it was increased to 128MB for 3.9 and beyond.
Updating PINN to 3.9.1 will not automatically resize any partitions. It tries to shoehorn the latest version into the same original 64MB partition by using a different wallpaper image. 

Since there is no wasted space on a PINN formatted drive, the RECOVERY partition can only be expanded by resizing and moving the partitions after it, which is a complicated procedure. Hence PINN will only resize the RECOVERY partition if you choose the `runinstaller` option which will also delete all your installed OSes. From 3.9 onwards, this will be a minimum size of 128MB. It maybe more if you have any additional files stored on your RECOVERY partition, like some OS installation files in the /os folder. PINN will also take account of the `reserve` configuration option which will allow the RECOVERY partition to be set to a specific size, or some additional space added to it.

The 3.9 versions only need about 64MB at the moment, which is why it can just about fit in the old partition size if the wallpaper image is swapped for a smaller one. But it doesn't leave any room for future expansion, which is why I set the minimum size to 128MB for future upgradability.

Please have a look at the initial post: https://forums.raspberrypi.com/viewtopic.php?p=2220339#p2220339 where I put some ideas of how you can adapt PINN for future upgradability. I would suggest using GParted to adjust your partition sizes. You cannot do this if any partitions on your SD card are mounted and actively being used, so you would need to put your SD card in a USB SD card adaptor and accesss it from another OS (e.g. a 2nd SD card running Raspberry PI OS) in which you have installed GParted. As long as you don't alter the order of the partitions, it will not affect PINN. However, this is a potentially dangerous operation, so it's a good idea to backup your SD card first if you have anything on it you don't want to lose.

lurch, 2024-09-08:
I guess it's more likely that some script inside one of the OSes that you run has tried to automatically enlarge the "boot partition" for that OS to 512MB, but has mistakenly assumed that the boot partition is `/dev/mmcblk0p1` which is actually the PINN partition?

procount, 2024-09-09:
With only an 8GB sd card, I suppose LibreELEC is the only OS installed?
If you are upgrading both PINN & LibreELEC for Pi5, maybe it's better to start afresh.

closed 2024-09-09 by rickwookie

procount, 2024-09-09:
PINN does not have any feature to delete OSes (because it has no ability to ADD OSes into the empty space that would be left).
The only way to do it properly within PINN is to backup the OSes you want to keep, then install the backups back again.

Deleting partitions manually can cause issues because partition numbers can change and break the PINN boot process.
It can be done if the partitions to be deleted are at the end of the drive as the partition numbers of the remaining OSes will remain the same. If not, you would have some fixing up to do, and this would require some in depth knowledge of how PINN works. I wouldn't recommend it.

rickwookie, 2024-09-09:
Thanks. I have backed up both OS's now to USB (after working out how to get my USB stick back to a single big FAT32 partition and create the required "os" folder. I had stupidly clicked the "use this for PINN, all data will be erased" prompt after selecting the USB drive from the bottom pull down menu in the main PINN page, rather than selecting it just from the drop-down in the 3rd (Maintenance) page).

I'll do a fresh PINN install now. That way I'll also get the more future-proofed 128 MB PINN partition, right? Is https://pinn.mjh.nz still a good way to get my partitions the size I want or is GParted a better way? Only, I note that it says there "Note: 100 MB will be allocated for the PINN software".

procount, 2024-09-09:
From v3.9, the PINN install menu has a new dialog to adjust the partition sizes itself, so there is no need to use pinn.mjh.nz