procount/pinn #838: OS's won't boot after fresh PINN install on RPi 5

rickwookie, 2024-09-09:
RPi 5 4GB
64 GB SD Card.
Unpacked pinn64 v3.9.2 from: https://sourceforge.net/projects/pinn/files/pinn64/pinn-lite.zip/download into a blank FAT32 formatted 64 GB SD Card.
replaced contents of cmdline.txt with:
`runinstaller quiet ramdisk_size=65536 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline no_group no_default_source select="allnetwork" disablesdimages alt_image_source=http://pinn.mjh.nz/os_list.json?libreelec_rpi5=8189&recalbox5=49311 loglevel=2 sdhci.debug_quirks2=4`
Took 5 attempts to get it to install initially (just kept getting stuck on "The install process will begin shortly" and had to reboot, but eventually it worked).
After installing the OS's (LibreELEC_RPI5 and recalbox5), neither will boot. They both just instantly reboot the Pi 5 and loop back to PINN.

procount, 2024-09-09:
1. It's better to use RPi Imager and install PINN as an image from the Misc Utility Categroy. It avoids unnecessary formatting of your SD card to FAT32, which is not always easy on very large SD cards.
2. You don't need to modify the cmdline.txt anymore for pinn.mjh.nz. You can set the partition sizes within PINN when you install the OSes.
3. Please see #770. The freezing is a known issue when the network connects, but I haven't found a solution yet. Please avoid using any USB peripherals (mouse/keyboard etc) until it has booted properly without freezing, and keep rebooting it if it does freeze.
4. If you are ending up in a boot loop, you probably have to update your eeprom firmware. See https://forums.raspberrypi.com/viewtopic.php?p=2173713#p2173713 for details, although probably updating to the very latest firmware version would be better now. This was due to a bug/omission in the RPi5 firmware.

rickwookie, 2024-09-09:
Yes, sorry, I've updated to the latest firmware now (rpi-boot-eeprom-recovery-2024-07-30-2712) and it's now working!

Re your point 

> 2. You don't need to modify the cmdline.txt anymore for pinn.mjh.nz. You can set the partition sizes within PINN when you install the OSes.

Yes, I noticed that new menu, and used it to use up the last few MB of space on my SD card so nothing was wasted. Does it also allow you to reorder the OS's or is there another easy way to do that that I've missed (I manually swapped os_list.json?libreelec_rpi5=8189&recalbox5=49311 from os_list.json?recalbox5=49311&libreelec_rpi5=8189 to achieve what I wanted).

procount, 2024-09-09:
You can reorder the OSes in the boot menu with the Up and Down buttons, although I have never understood why people want to do this.

rickwookie, 2024-09-09:
> You can reorder the OSes in the boot menu with the Up and Down buttons, although I have never understood why people want to do this.

lol, no not for neatness in the boot menu (or alphabetical sorting or whatever). I mean the actually order that they are installed (partition order) on the SD card. I specifically may want certain os's to be any the end of the drive, to make playing with their partition size and the free-space easier, and put os's that I don't expect to have to tweak their size before them on the drive (so I can resize the others without having to move the position of those "fixed" ones)

procount, 2024-09-09:
This was never a design requirement of NOOBS/PINN.
The following will only work if your OSes are all of the same 2 partition standard layout (like Raspios).
Instead of installing your OSes directly, install a number of projectspaces first. It requires a bit of preplanning since you need to set their partition sizes according to the OSes you want to install in the correct order.
Once they are installed, clear your selections, then select the list of OSes you want to install. Go to the maintenance menu, select all the Project Spaces and select Replace. Then mathc up the projectspaces with the OSes you want to install at each position.
It's a bit longwinded, but should work.

rickwookie, 2024-09-09:
It's certainly long winded. I'm tempted to stick to the method I used tbh.

Just a couple of final issues. Is there a list of outstanding things not yet working on the Pi 5? CEC doesn't seem to be working yet for example, and I haven't tried joystick control yet.
Also, now I have it all booting on the Pi 5, how do I restore to backup I made earlier of my old LibreELEC OS? If I put the USB stick in, and select "replace" on the new LibreELEC install, I don't see how to choose the backup to replace it with.

procount, 2024-09-09:
>  Is there a list of outstanding things not yet working on the Pi 5?

No definitive list yet, but anything related to the display (KMS, TVservice, CEC, Hyperpixel etc) Joysticks, Network causing GUI freeze, wifi adaptors are mostly affected. Basically, most enhancements over and above what NOOBS did since they were mostly related to the VC4 graphics in some way that is now done via KMS instead.

> What's with the tiny font though for the window title text on the Pi 5?

Something to do with KMS. The title font is now managed by the System rather than the Application and I haven't found out how to control it. Originally it affected ALL fonts (on some displays) which is now fixed, apart from this last one.