procount/pinn #839: Raspbian Only App to Boot w/ official 7" touchscreen

Asherk123, 2024-09-12:
**Describe the bug**
Complete tech noob trying to setup a cyberdeck with PINN. Loaded PINN via pi imager downloaded Kali Linus full, raspbian full, and aos13 all rpi4 models. Upon boot only raspbian loads, Kali gives black screen, AOS (also tried with retropie) give white screen fades to black with white lines. Using rpi4 official 7" touch display with display cable, I have a USB speaker and USB keyboard plugged into device both drawing some additional power. 

**To reproduce**
Download PINN on SD card with Pi imager
Install Kali Linux full, raspbian full, and aos13
Boot Kali Linux (black screen)
Boot raspbian (loads and functions as expected)
Boot on AOS (screen goes white then fades black)

**Actual behaviour**
Kali and AOS not loading on boot

**System**
Add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4
* Which version of PINN?
3.9

**Logs**
If applicable, add the relevant output from `/tmp/debug` or `dmesg`.

**Additional context**
First time ever posting on git so if this is the wrong place to ask for advice feel free to remove! Thanks for the sweet program!

procount, 2024-09-12:
> Kali and AOS not loading on boot

Well, they are probably loading, just not displaying on the correct screen. Have you tried plugging a HDMI monitor in? You may need to do this anyway in order to configure the OS to use the DSI touchscreen (see below).

Raspios does a good job of auto-detecting the DSI display and loading up the correct drivers.
With other distros, you may have to configure them manually to use the DSI display.
How you do this depends on the OS and whether it uses KMS display drivers or not.
For AOSP13, please read KonstaKang's FAQ on:  https://konstakang.com/devices/rpi4/AOSP13/

Some details for configuring KMS mode can be found here: https://www.raspberrypi.com/documentation/computers/configuration.html#set-the-kms-display-mode
You may need to add an appropriate dtoverlay in config.txt to load the appropriate DSI driver.

Some Legacy (non-kms) options can be found here: https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#lcd-displays-and-touchscreens

Please remember that in PINN's multiboot environment, each OS is completely separate with it's own boot partition and corresponding configuration settings, firmware & kernel etc. So each OS needs to be configured separately.

procount, 2024-09-13:
I've just installed Kali64, AOSP13 and Raspberry PI OS on my PI4 with the 7" touchscreen.
Kali worked out on the touchscreen out of the box - no configuration was required.
AOSP first has to be booted on HDMI, then select Settings -> System -> Raspberry Pi settings -> Touchscreen and enable it. You need to disconnect any HDMI display when using the DSI display. Then AOSP worked as well. Screen blanking happens after one minute. You need to press the power key (F5) to turn it on again. Not sure what you do if you don't have a keyboard, so might be a good idea to increase the screen blanking time.

Asherk123, 2024-09-13:
Unbelievably helpful thank you so much I'll be attempting this via HDMI
after work. Thanks again!