procount/pinn #841: Booting into shell or bootloader and not OS

Dabuke, 2024-09-15:
Describe the bug
When booting from NVMe into PINN and pressing shift to select an OS, either Raspios or Kali, it ends up ether in the bootloader or being dropped directly into a busybox shell. (Raspios into busybox, Kali into bootloader) and not loading either OS to continue normal operation.

To reproduce
Start up Raspios from an SD card and launch Imager, select Pi5, select PINN and then select the NVMe to install PINN onto.

Once that has correctly formatted the NVMe and installed PINN change the boot order to NVMe using a terminal and the command sudo raspi-config now shutdown the Pi5.

Remove the SD card and power on the Pi5

When PINN launches select go through the steps to install the following OSs Raspios arm64 full version and the latest Kali OS, select the NVME to install onto. Once that is finished setting up and installing let it launch into either OS. Reboot the Pi5 a second time and try to boot into either OS.

Expected behaviour
It is expected either OS would launch into the login screen.

Actual behaviour
If selecting Raspios your get dropped into the busy box shell.

If you select Kali you get dropped into the bootloader.

Both end up stuck in these situations until you power off the Pi5.

System
Add answers to the following questions:

Model of Pi = Pi5
PINN Version = 3.9.2
Logs
Screenshots
[image]
[image]
[image]
[image]

procount, 2024-09-15:
This looks like an issue with your NVMe SSD as the root filesystem cannot be loaded/found.
You didn't mention which NVMe SSD or HAT you are using.
Have you followed the instructions on how to install and enable it?
Have you already managed to install an OS directly to your SSD without PINN?

You can enable SSD through config.txt, but you would have to do this for EACH OS you install.

According to Jeff Geerling's noted you should do the following:

```# Edit the EEPROM on the Raspberry Pi 5.
sudo rpi-eeprom-config --edit

# Change the BOOT_ORDER line to the following:
BOOT_ORDER=0xf416

# Add the following line if using a non-HAT+ adapter:
PCIE_PROBE=1

# Press Ctrl-O, then enter, to write the change to the file.
# Press Ctrl-X to exit nano (the editor).
```
Hopefully that should enable the SSD for each OS.

Dabuke, 2024-09-16:
Hi Procount,

Thanks for getting back to me, and thanks for all the work you do across all the platforms responding to people.

From all the troubleshooting I done before submitting this post I am unable to remember if I done anything specific for the SSD. However, yes I did manage to get Kali installed and working reliably on the SSD when it was the only OS running on it. But I didn't like the thought of using that as my only OS for research and writing in case something happened and I lost my work.

I a using a Sabrent Rocket NVMe PCLe 4.0 256GB Model: SB-2130-256 (M.2 2230). No hat as the SSD is plugged in directly using the inbuilt PCLe expansion connector.

I have made the change to the rpi-eeprom-config file as you previously suggested when booted into Pi OS, and so far have been able to shutdown and launch into it successfully multiple times, however, I am unable to launch into Kali from PINN. Also during this testing I have not connected to WiFi and I have skipped all the updates during setup. I will do all this and update here.

Dabuke, 2024-09-16:
Sorry, thank you for prompting me.

I have an Argon NEO 5 NVMe EB.

OK, so sees to not be an issue running Kali from PINN as its own, is there anything specific I need to do if I am dual booting either Raspios or Kali?

Due to using the Argon EB do I need to add the `PCIE_PROBE=1` as you mentioned?

procount, 2024-09-16:
> Due to using the Argon EB do I need to add the PCIE_PROBE=1 as you mentioned?

Yes. I assumed you had already done this. See https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#boot-from-pcie

You may want to start reading from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-connector-for-pcie

Dabuke, 2024-09-29:
I would like to say this is fix, I am fairly sure it is as I have been using my Pi and switching between the two OSs.

Thanks Procount and thanks for all the time you put into helping people like me.

Take care!

Dabuke, 2024-09-29:
Ran through the steps stated above.

closed 2024-09-29 by Dabuke