procount/pinn #843: headless setup, impossible: recovery.cmdline not found at all, autoboot.txt not working

magowiz, 2024-10-06:
**Describe the bug**

Preface: For me this is not necessarily a bug but a lack in documentation, the fact that I didn't understand where to put those files, doesn't mean necessarily is there a bug.

No viable way to control PINN booting system without keyboard in current setup, I installed PINN via raspberry imager on Sd card, I successfully created a dual boot with LibreElec and Android, I read the readme talking about solutions regarding autoboot.txt (to autoselect at boot writing a file) and recovery.cmdline but I cannot find these files at all across sd partitions and for me is not clearly explained which is PINN directory, since I don't have anything called like this, could you please address me which is the right place to put those file, if I need to create from scratch?  

I created autoboot.txt in every partition, but nothing changes.

**To reproduce**
1. install PINN using raspberry imager
2. install libreelec and android as operating systems
3. create a autoboot.txt with a valid partition number and put it on root of each partition

**Expected behaviour**
At least one partition should be good for it and autoboot.txt behavior should be in place, skipping PINN menu, also recovery.cmdline or some examples about it should exist in order to understand where to put it, to try enable ssh and/or vnc

**Actual behaviour**
file autoboot.txt is ignored every time

**System**
Add answers to the following questions:

* Which model of Raspberry Pi? Raspberry Pi 4 model B
* Which version of PINN? 3.9.2

procount, 2024-10-06:
Apologies if the documentation is not clear enough. This is compounded by the changes required for the Pi5 that haven't percolated into all parts of the documentation yet.

Firstly let me address the general questions you raise, and then I'll tackle your specific solution.

> headless setup, impossible: recovery.cmdline not found at all

> No viable way to control PINN booting system without keyboard in current setup

PINN is a continuation of NOOBS which only had a GUI interface with a keyboard and mouse interface. With its additional features, PINN can now be controlled with a touchscreen, mouse or joystick alone using the onscreen keyboard. It can also be controlled headlessly over a network using a VNC client. Access is also provided over SSH, although this does not provide access to the GUI controls and is more suited to maintenance activities such as editing files etc.

So headless setup, which normally refers to the lack of a monitor and maybe keyboard/mouse as well, leaving just network access available, is entirely possible. It has also been shown possible to install PINN onto a remote Pi that had an existing Raspbian OS installation and then install multiple OSes onto the device using PINN. 

{*NOTE not all of the above features are available on the Pi5 yet due to the change in architecture, but they are being worked on).

NOOBS/PINN used to use the "alternate load mechanism" for the kernel, which meant most of its boot/firmware files were renamed to include "Recovery" or an abbreviation of that in their filename. With the advent of the Pi5, this mechanism was dropped, so PINN had to revert to the more usual names used in a typical OS installation. For this reason, `recovery.cmdline` has been renamed to the more usual `cmdline.txt`.

Rather than editing these file manually, cmdline.txt and config.txt are editable within PINN from the Maintenance Menu. THis applies to all OSes, including PINN.

`Autoboot.txt` is not a PINN specific file, but rather something that the firmware will read on startup. It does not normally exist and needs to be created if you want to use it. You can find details about it on Raspberry Pi's documentation webpages e.g. https://www.raspberrypi.com/documentation/computers/config_txt.html#autoboot-txt. 

It is important to realise that autoboot.txt is only read from the initial boot partition of your boot drive. So if you are using PINN, this means PINN's boot partition (usually /dev/mmcblk0p1 if using an SD card) which is labelled RECOVERY. Since it has this specific name, most other OSes will refuse to mount it automatically, so you will not find it in LibreELEC or Android etc. unless you specifically mount that partition.

If you want to access PINN's RECOVERY partition, the easiest way is to remove the SD card from the PI and put it in a USB adaptor inserted into a WIndows PC. The RECOVERY partition will most likely be the only partition that will be visible to Windows as the remainder will not be understood by Windows (DO NOT let Windows reformat anything - Just because it doesn't understand the Linux partitions does not mean they are faulty!)
You can now add or edit the autoboot.txt file using Windows Notepad or similar.

You can also access PINN's recovery partition from the shell within PINN. Just press ctrl-alt-F2 to access the shell and login with user=root, password=raspberry. The recovery partition is mounted on /mnt, but it is normally read-only. In order to edit anything on it, use the command: `mount -o remount,rw /mnt` and then `cd /mnt`. You can use the included `nano` editor to edit autoboot.txt.

Note that autoboot.txt can include conditional statements, so it could be used to boot a different OS based on the state of some GPIO pins, for example.  You might also want to read about the Pi's tryboot feature.
If you use this autoboot.txt file, you would need to delete the autoboot.txt file to access PINN again.

If you do have some other input devices (e.g. mouse, joystick etc) then you could also consider using PINN's stickyboot method, especially if you use one OS most of the time and only occasionally want to choose the other.

mrengy, 2025-08-31:
Please update references to `recovery.cmdline ` in the [main readme](https://github.com/procount/pinn/blob/master/README_PINN.md). It's pretty time consuming to try to find a non-existent file and track down why it's not showing up each time I try to change a configuration option.

closed 2025-09-01 by procount