procount/pinn #847: Disk merging during system exchange

JanSykor, 2024-10-29:
Heslo @procount 
Is there some function in Pinn which allowing partition merging.of installed system (root and boot) before system exchange and disk partitioning depend new system instalation?
My problem is that I installed in the past LibreElec instead of your spare system and I have not enough space on boot partition for update of LibreElec now.
Thanks for your help 
Jan

procount, 2024-10-30:
Not within PINN, no.

Merging partitions is not a good idea because it changes all the partition numbers after the merge. If you do this, there is a good chance that PINN will not be able to find any of its OS boot partitions, and even if it could, each OS may not be able to find any of its constituent partitions without a  lot of manual modifications.

A better solution would be to resize your partitions, as this does not change the partition order.
To do this, you would need to use a tool like GParted on another OS, as you cannot modify mounted partitions. Preferably an OS on a different drive or computer, and attach your PINN drive as a secondary drive (say in a USB SD card reader) to avoid any possible problems.

With GParted you could shrink the size of your LibreElec root partition and then expand the boot partition by the same amount, to give you enough room for the new version. If your root partition is not big enough to do this, you would need to do some similar shuffling with the partitions from other OSes to move the free space to where you need it.

JanSykor, 2024-10-30:
Thanks for your help
I did not expected problem with partition name.
I use Gparted and try resize both partition.
Next think what I dont know, was that final size of FAT16 boot partition must be multiply of 32. If not, Gparted want change format to FAT32 and this destroyed data on boot partition.
Finaly I changed size to 512 MB (32x16), reformat to FAT16 and reinstal LibreElec in Pinn
After applying of LibreElec system backup is now all  working.
Do you know if exist some similar rules for ext4 partition formát?
Jan

JanSykor, 2024-10-30:
BTW
Is there any possibilities how to use for resizing your new partition tool in Pinn?
Thanks 
Jan

procount, 2024-10-30:
> I did not expected problem with partition name.

There are at least 4 ways of referencing a partition, and each OS uses different ones, so my answer played safe for the general case. But it also depends on the referencing method of ALL the following OSes not just the one you want to merge.

> Do you know if exist some similar rules for ext4 partition formát?

No I'm not an expert in the rules.

> Is there any possibilities how to use for resizing your new partition tool in Pinn?

As you have discovered, resizing partitions can be complicated to get right, which is why PINN shies away from it 😉 

Glad you got it sorted though. Feel free to close this issue if it is now solved.