procount/pinn #849: Update Kernel

procount, 2024-11-02:
* Raspberry Pi firmware a2e586ba98ce68f7d11b1c717ad8329b95dcb3b6
  * Linux kernel 6.6.51 - 5aeecea9f4a45248bcf564dec924965e066a7bfd

NumberOneGit, 2024-12-16:
The new CM5 units are finally shipping and my 4GB Ram, 32GB eMMC is posting the same Async error as the 2GB pi5 units when trying to run PINN.  Do you have a beta version of the 3.9.3 update available?

NumberOneGit, 2024-12-16:
Great!  Thanks for all the work you put in

NumberOneGit, 2024-12-19:
So far, I have been able to install PINN on an NVME attached to the CM4 with 3.9.2 installed and perform the update, then switch the CM4 out with a CM5.   Does the current version of PINN not have DWC2?   RPOS is using otg_mode=1 for CM4, but DWC2 for CM5 and as of now I can’t get any USB to work, I can only access via SSH or VNC.  I’m pretty sure previous versions of PINN had DWC2, so maybe it’s a different USB issue.  The first version of RPOS that works with CM5 is November’s and PINN is still on the October version.   AOSP and Lineage work but only the versions that were released a few days ago.   For some reason, I can’t get 3.9.3 to work at all on the eMMC of the CM5 only on the NVME I brought over from the CM4.  When using that NVME along with the CM5, the eMMC doesn’t show up as a second drive option.

procount, 2024-12-19:
Thanks for reporting. Unfortunately, I don't have a CM5 to test with, but I'll look into it.
There are plans to change the build architecture of PINN in the near future, which should eliminate such issues, but for now I'll see if I can add what's missing.

NumberOneGit, 2024-12-23:
My official Raspberry Pi CM5 board just arrived and I will revise this to say that I don't think DWC2 is necessary.  Older CM4 boards work with the CM5, but boards specifically made for the CM5 should have USB3 and not need the drivers.  I will try installing PINN directly on the CM5 again later this week to see if it's able to boot with the official carrier board.    Many of the OSes on PINN will need to be updated to newer versions as the CPU stepping on the CM5 and the Pi5 2GB is different than on the original Pi 5.

NumberOneGit, 2025-01-27:
So I have been using PINN on the CM5 and for the most part it seems to be operating the same way that it does with the Pi 5 (some freezing up but workable).  However, this is only with NVME over PCIE or in a USB enclosure or another type of USB storage.   
I am still unable to boot the eMMC with PINN installed on it or even have the eMMC recognized by PINN when booting from another drive.   I have used the same eMMC to boot other OS’s with no issue, so it’s not a problem with my device.   No issues at all with any of the newer versions of PINN booting from my CM4 eMMC.

NumberOneGit, 2025-02-25:
It may just be an issue of updating the kernel.   I had the same issue with yocto/meta-raspberry pi on the CM5 and updating to a newer kernel solved it.   I’ll wait for the next kernel update and will report back then.

procount, 2025-02-25:
You could try installing pinn v3.9.4 manually from my github releases. It may still not work, but maybe worth a try.